RavenSystem/esp-homekit-devices #651: Shelly1 PIR motion sensor

Kienz, 2019-12-09:
I want to use a Shelly 1 to make a standard PIR motion sensor smart. There should be two devices in HomeKit - motion sensor and toggle switch.
eg. https://gist.github.com/xbmcnut/6d239de9523246d7a831ca55be6564c2

The motion sensor and the toggle switch should be seperated - the light is turned on by an HomeKit automation.

Is this possible? If yes, what would the JSON look like?

mafyata, 2019-12-09:
Try this:
 {"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"t":1,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":12,"i":10,"f1":[{"g":14}]}]}

Just check which gpio will you use. Here is for gpio14

Kienz, 2019-12-09:
@mafyata How can I use gpio14 on Shelly1. I'm new to this and I don't know how to use the gpio's. How do I wire the PIR motion sensor. Equal to the picture form https://gist.github.com/xbmcnut/6d239de9523246d7a831ca55be6564c2?

mafyata, 2019-12-09:
I don’t know shelly’s gpio. I don’t use shelly.

mafyata, 2019-12-09:
I use sonoff devices

Kienz, 2019-12-10:
@RavenSystem Is the following json for Shelly 1 correct?
Toggle Switch on I/O and Motion Sensor on SW.
[image]

```json
{"a":[{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"t":1,"b":[],"s":0},{"t":12,"i":10,"f1":[{"g":5}],"s":0}]}
```

RavenSystem, 2019-12-10:
Shelly devices must disable pull-up resistor for digital inputs ("p":0):

```json
{"a":[{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]}},{"t":12,"i":10,"f1":[{"g":5,"p":0}]}]}
```

Kienz, 2019-12-10:
With this config I don’t get 240V on I/O if toggle is on. What is wrong?

closed 2019-12-18 by Kienz