RavenSystem/esp-homekit-devices #652: Documentation changes

WizBangCrash, 2019-12-09:
@RavenSystem I have been updating all my Sonoff's from RavenCore to HAA and it has been quite a learning curve getting to understand the JSON Config.
As I have tried to learn more about HAA I have been updating my own copy of the Wiki documentation to capture these things.
Recently I got carried away and have completely reformatted / restructured the documentation to help me navigate to the bits I want faster.
Would you be interested in merging these changes into your own Wiki?
If yes, let me know and we can sort out the details of how to merge.

My wiki can be found here: https://github.com/WizBangCrash/esp-homekit-devices/wiki

It's not complete yet, but is probably 80% there. I plan to add more examples once I understand a little more about the "f0" specific status buttons :-)

You're doing great job BTW. Keep up those improvements and new features coming.

RavenSystem, 2019-12-10:
It is amazing!! A great job! Of course, I am interested. Do you want to be collaborator of this wiki repo?

WizBangCrash, 2019-12-11:
I'm happy to be a collaborator if you are okay with that.
I'll finish off changing everything over to the new format and then I'm happy to merge it into the main repo.

RavenSystem, 2019-12-11:
Check: https://github.com/RavenSystem/esp-homekit-devices/invitations

WizBangCrash, 2019-12-12:
@RavenSystem What's your preference. Do you want me to merge in my current changes to master and then continue to edit and update this repo's Wiki as I modify it, or would you prefer me to merge my changes into a wiki repo branch for you to review etc. before merging into master?

I've still got a fair bit to do, but I think the facts are correct, so I don't think my updates will mislead people (too badly :-)

RavenSystem, 2019-12-13:
Merge to master. 

BTW: Are you on Discord chat?

WizBangCrash, 2019-12-13:
Okay. Give me a couple more days to clean things up and then I'll merge.
I have a couple of images you will probably want to move to your media repo when I do.

No. I don't use Discord.

RavenSystem, 2019-12-13:
I have added you to media repo too: https://github.com/RavenSystem/ravensystem-media/invitations

I ask about Discord to talk privately. Another way? iMessage?

St0nehead, 2019-12-13:
When i tried to follow your link, i always getting error:

Sorry, we couldn't find that repository invitation. It is possible that the invitation was revoked or that you are not logged into the invited account.

St0nehead, 2019-12-13:
iMessage, WhatsApp: +77001999499

closed 2019-12-16 by RavenSystem