RavenSystem/esp-homekit-devices #654: FTDI missing on macOS?

brymonster, 2019-12-10:
Hey all! Trying to flash my newly arrived magic home (ZJ-WFMN-A V1.1) with HAA on macOS 10.15.1
I believe I have it all connected properly, and have flashed devices with the same FTDI and Mac before. Now Im getting the below response, with the FDTI connected or NOT.
Any help would be appreciated!

```
$ esptool.py -p /dev/tty.usbserial-A50285BI erase_flash
esptool.py v2.8-dev
Serial port /dev/tty.usbserial-A50285BI
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 4, in <module>
    __import__('pkg_resources').run_script('esptool==2.8.dev0', 'esptool.py')
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 666, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 1453, in run_script
    exec(code, namespace, namespace)
  File "/Library/Python/2.7/site-packages/esptool-2.8.dev0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 3201, in <module>
    _main()
  File "/Library/Python/2.7/site-packages/esptool-2.8.dev0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 3194, in _main
    main()
  File "/Library/Python/2.7/site-packages/esptool-2.8.dev0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 2883, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "/Library/Python/2.7/site-packages/esptool-2.8.dev0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 273, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "/Library/Python/2.7/site-packages/esptool-2.8.dev0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 237, in __init__
    self._port = serial.serial_for_url(port)
  File "/Users/XXXX/Library/Python/2.7/lib/python/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/Users/XXXX/Library/Python/2.7/lib/python/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/tty.usbserial-A50285BI: [Errno 2] No such file or directory: '/dev/tty.usbserial-A50285BI'
```

System Reports sees the FDTI with the same serial I have used in the past successfully:

[image]

```
$ ls /dev
afsc_type5			ptysa				ttypd
auditpipe			ptysb				ttype
auditsessions			ptysc				ttypf
autofs				ptysd				ttyq0
autofs_control			ptyse				ttyq1
autofs_homedirmounter		ptysf				ttyq2
autofs_notrigger		ptyt0				ttyq3
autofs_nowait			ptyt1				ttyq4
bpf0				ptyt2				ttyq5
bpf1				ptyt3				ttyq6
bpf2				ptyt4				ttyq7
bpf3				ptyt5				ttyq8
bpf4				ptyt6				ttyq9
console				ptyt7				ttyqa
cu.Bluetooth-Incoming-Port	ptyt8				ttyqb
disk0				ptyt9				ttyqc
disk0s1				ptyta				ttyqd
disk0s2				ptytb				ttyqe
disk1				ptytc				ttyqf
disk1s1				ptytd				ttyr0
disk1s2				ptyte				ttyr1
disk2				ptytf				ttyr2
disk2s1				ptyu0				ttyr3
disk2s2				ptyu1				ttyr4
disk2s3				ptyu2				ttyr5
disk2s4				ptyu3				ttyr6
disk2s5				ptyu4				ttyr7
disk3				ptyu5				ttyr8
disk3s1				ptyu6				ttyr9
disk3s2				ptyu7				ttyra
disk4				ptyu8				ttyrb
disk4s1				ptyu9				ttyrc
disk4s2				ptyua				ttyrd
disk5				ptyub				ttyre
disk5s1				ptyuc				ttyrf
disk5s2				ptyud				ttys0
disk6				ptyue				ttys000
disk6s1				ptyuf				ttys1
disk7				ptyv0				ttys2
disk7s1				ptyv1				ttys3
disk7s2				ptyv2				ttys4
disk7s3				ptyv3				ttys5
disk8				ptyv4				ttys6
dtrace				ptyv5				ttys7
dtracehelper			ptyv6				ttys8
fbt				ptyv7				ttys9
fd				ptyv8				ttysa
fsevents			ptyv9				ttysb
io8log				ptyva				ttysc
io8logmt			ptyvb				ttysd
io8logtemp			ptyvc				ttyse
klog				ptyvd				ttysf
lockstat			ptyve				ttyt0
machtrace			ptyvf				ttyt1
null				ptyw0				ttyt2
oslog				ptyw1				ttyt3
oslog_stream			ptyw2				ttyt4
pf				ptyw3				ttyt5
pfm				ptyw4				ttyt6
profile				ptyw5				ttyt7
ptmx				ptyw6				ttyt8
ptyp0				ptyw7				ttyt9
ptyp1				ptyw8				ttyta
ptyp2				ptyw9				ttytb
ptyp3				ptywa				ttytc
ptyp4				ptywb				ttytd
ptyp5				ptywc				ttyte
ptyp6				ptywd				ttytf
ptyp7				ptywe				ttyu0
ptyp8				ptywf				ttyu1
ptyp9				random				ttyu2
ptypa				rdisk0				ttyu3
ptypb				rdisk0s1			ttyu4
ptypc				rdisk0s2			ttyu5
ptypd				rdisk1				ttyu6
ptype				rdisk1s1			ttyu7
ptypf				rdisk1s2			ttyu8
ptyq0				rdisk2				ttyu9
ptyq1				rdisk2s1			ttyua
ptyq2				rdisk2s2			ttyub
ptyq3				rdisk2s3			ttyuc
ptyq4				rdisk2s4			ttyud
ptyq5				rdisk2s5			ttyue
ptyq6				rdisk3				ttyuf
ptyq7				rdisk3s1			ttyv0
ptyq8				rdisk3s2			ttyv1
ptyq9				rdisk4				ttyv2
ptyqa				rdisk4s1			ttyv3
ptyqb				rdisk4s2			ttyv4
ptyqc				rdisk5				ttyv5
ptyqd				rdisk5s1			ttyv6
ptyqe				rdisk5s2			ttyv7
ptyqf				rdisk6				ttyv8
ptyr0				rdisk6s1			ttyv9
ptyr1				rdisk7				ttyva
ptyr2				rdisk7s1			ttyvb
ptyr3				rdisk7s2			ttyvc
ptyr4				rdisk7s3			ttyvd
ptyr5				rdisk8				ttyve
ptyr6				sdt				ttyvf
ptyr7				stderr				ttyw0
ptyr8				stdin				ttyw1
ptyr9				stdout				ttyw2
ptyra				systrace			ttyw3
ptyrb				tty				ttyw4
ptyrc				tty.Bluetooth-Incoming-Port	ttyw5
ptyrd				ttyp0				ttyw6
ptyre				ttyp1				ttyw7
ptyrf				ttyp2				ttyw8
ptys0				ttyp3				ttyw9
ptys1				ttyp4				ttywa
ptys2				ttyp5				ttywb
ptys3				ttyp6				ttywc
ptys4				ttyp7				ttywd
ptys5				ttyp8				ttywe
ptys6				ttyp9				ttywf
ptys7				ttypa				urandom
ptys8				ttypb				xcpm
ptys9				ttypc				zero
```

mafyata, 2019-12-10:
Magic home led controller have 2 GRN. You must use one on the side where is tx,rx. And must me connected with IO0 to enter in flash mode.

brymonster, 2019-12-10:
@mafyata thank u for the reply. 
I did that, as seen [here.](https://user-images.githubusercontent.com/192647/48788046-006bc300-ecb8-11e8-9e0e-58145b374778.jpeg)
Unfortunately, I get the same response in terminal every time, whether the FTDI is connected to not, leading me to believe my computer is no longer "seeing" the device.
Ideas?

mafyata, 2019-12-10:
Nope. I flashed my devices with one GRN , not 2. Where you connected grn to io0 is mistake. I use only the grn that you connected to ftdi.

brymonster, 2019-12-10:
So you are saying I should connect io0 to the same ⏚ as FTDI when powering up?

mafyata, 2019-12-10:
You must connect 3v too.
[image]

mafyata, 2019-12-10:
Yes , same grn. It can be connected on all time of flashing , not only when powering up.

brymonster, 2019-12-10:
Hooked up as recommended, same exact error:
```brymonster$ esptool.py -p /dev/tty.usbserial-A50285BI erase_flash
esptool.py v2.8-dev
Serial port /dev/tty.usbserial-A50285BI
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 4, in <module>
    __import__('pkg_resources').run_script('esptool==2.8.dev0', 'esptool.py')
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 666, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 1453, in run_script
    exec(code, namespace, namespace)
  File "/Library/Python/2.7/site-packages/esptool-2.8.dev0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 3201, in <module>
    _main()
  File "/Library/Python/2.7/site-packages/esptool-2.8.dev0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 3194, in _main
    main()
  File "/Library/Python/2.7/site-packages/esptool-2.8.dev0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 2883, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "/Library/Python/2.7/site-packages/esptool-2.8.dev0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 273, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "/Library/Python/2.7/site-packages/esptool-2.8.dev0-py2.7.egg/EGG-INFO/scripts/esptool.py", line 237, in __init__
    self._port = serial.serial_for_url(port)
  File "/Users/brymonster/Library/Python/2.7/lib/python/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/Users/brymonster/Library/Python/2.7/lib/python/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/tty.usbserial-A50285BI: [Errno 2] No such file or directory: '/dev/tty.usbserial-A50285BI'```

mafyata, 2019-12-10:
Is it plugged to 12v ?

brymonster, 2019-12-10:
No. I thought it was EITHER 12v or 3v3 from FTDI. Not both. 
I have tried it both ways. Neither with success. I do believe, as the issue title states, its a problem with my computer recognizing the USB serial FDTI, not with how its hooked up to the board.

mafyata, 2019-12-10:
You must connect 12v to magic home plug and 3v to ftdi. When connect 12v then connect ftdi to usb.

brymonster, 2019-12-10:
Same issue. 
I need help diagnosing why the FDTI isn't responding on my Mac suddenly. 
Its not a hardware connection issue.

mafyata, 2019-12-10:
I can’t help. I use windows and can see if ftdi is recognized and which com port is using.

brymonster, 2019-12-10:
Im getting closer!
For any of those with macOS on 10.15.1 or above..
I deleted the `FTDIKext.kext` from `/Library/Extensions` and restarted. 
Now my FDTI is working back to normal! Show up under `ls /dev`.
But I now im stuck getting `Connecting.....-----.....-----.....----- `
I must not have it wired properly. 

- I have tried with and without 12v connected. Also, with and without VCC connected, since I have found conflicting arguments.

- i have tried momentarily jumping io0 to GND on the FTDI while connecting, and also permanently connecting there as well. nothing seems to work.  

Any help is welcome!

Heres my ZJ-WFMN-A V1.1 wiring:
[image]

RavenSystem, 2019-12-10:
Remove 12V!!! And try swapping TX and RX.

brymonster, 2019-12-10:
No good. 
I will try redoing some of my solders.

mafyata, 2019-12-10:
It will not flash if it don’t have 12v. 
.....——-.......——- is showing when magic home is not in flash mode. When i flashed mine devices I have connected 12v, then plugged ftdi. Grn and io0 were connected on all the time.

brymonster, 2019-12-11:
Solved!
Thank you both for the support. 
Swapping RX/TX didnt help. All my connections had continuity, so I just kept trying different power io0 to ⏚ combos.

@mafyata, mine worked **without 12v**. I also attached io0 to FDTI ⏚ while connecting, then detached after 3 seconds. 

I will be attempting to use [this](https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#magichome-rgbw-led-strip-controller-r-pin-15-g-pin-13-b-pin-12-w-pin-14) template for this ZJ-WFMN-A V1.1 device.

brymonster, 2019-12-11:
### 
Using [this](https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#magichome-rgbw-led-strip-controller-r-pin-15-g-pin-13-b-pin-12-w-pin-14) template I cannot get the GREEN LED to turn on at all. And all the colors are shifted. 

Also the below JSON as suggested [here](https://github.com/RavenSystem/esp-homekit-devices/issues/476) on the same device as mine. But wouldn't kick out of setup mode. 

```
{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":12,”g”:5,”v”:13,”w”:15,”b”:[{“g”:0}]}]}
```
Suggestions?

mafyata, 2019-12-11:
"  ”
Did you see differences?

mafyata, 2019-12-11:
You must use this " , not that ”

brymonster, 2019-12-11:
oh snap!
I was using the notes app to adjust the JSON, and it corrected the " to ”. 
Did not even realize. thanks again @mafyata 
I will try it now.

mafyata, 2019-12-11:
Not all magic home devices are same. I have 2 rgb controllers. 
One with this json: {"c":{"n":"Kuhnq-Led","b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13}]}

And one with this: {"c":{"n":"Hol-Led","b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13,"fr":5,"fg":5,"fv":5}]}

brymonster, 2019-12-11:
```
{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":12,"g":5,"v":13,"w":15,"b":[{"g":0}]}]}
```
Works beautifully with RGBWW. thank you!

closed 2019-12-11 by brymonster