RavenSystem/esp-homekit-devices #655: JSON help - Garage Door

yusufd1, 2019-12-10:
Hi please can someone help me with a json. It will be on a Wemos D1 Mini and requires the following:
- Relay on GPIO 12 that inches for 0.6 seconds to open or close garage door
- Close Sensor on GPIO 13 
- Open Sensor on GPIO 14
- if door is opening/closing and home app button pressed then GPIO 12 must fire twice so door stops and changes direction [I think this is called stop step]
- the normal time for door to open or close is  15 seconds 

Thanks for the help.

RavenSystem, 2019-12-10:
Try this, but I don't know if your sensors are NO (normal open) or NC (normal close):

`{"a":[{"t":40,"d":15,"f3":[{"g":13,"t":0}],"f4":[{"g":13},"f2":[{"g":14,"t":0}],"f5":[{"g":14}],"0":{"r":[{"g":12,"v":1,"i":0.6}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.2},{"g":12,"v":1,"i":0.6},{"g":12,"v":1,"i":1.8}]},"3":{"a":2}}]}`

yusufd1, 2019-12-11:
I will try it. Is it correct I saying that no and no will just change the “t” in the GPIO 13 and 14 to either 0 or 1. If that’s the case then I can test it and use trial and error with the json?

yusufd1, 2019-12-11:
I will try it. Is it correct I saying that no and nc will just change the “t” in the GPIO 13 and 14 to either 0 or 1. If that’s the case then I can test it and use trial and error with the json?

yusufd1, 2019-12-12:
@RavenSystem please help. Having problems with the json. The garage door worked perfectly on the below pictured RC settings. What will be the json for these exact settings?
[image]
[image]

yusufd1, 2019-12-12:
As you can see reverse switch is on and while open sensor is off in these settings I used to have it on 2 when I had the open sensor connected which I do wish to have in HAA.

closed 2020-01-10 by RavenSystem