RavenSystem/esp-homekit-devices #656: Relay 8CH

skv6, 2019-12-10:
I tried make relay for 8ch. It work for 7 max and not more. When i made 8 - homekit cant see device. Has it limit?

RavenSystem, 2019-12-10:
What is your JSON?

skv6, 2019-12-11:
7CH   I can add it to HomeKit. Usually it doesnt respond after, bu homekit can see it
{«c":{"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":2}]},"1":{"r":[{"g":2,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":1,"b":[],"s":1},{"0":{"r":[{"g":1,"v":0}]},"1":{"r":[{"g":1,"v":1}]},"t":1,"b":[],"s":1},{"0":{"r":[{"g":3,"v":0}]},"1":{"r":[{"g":3,"v":1}]},"t":1,"b":[],"s":0}]}

8CH   Cant add it to homekit
{"c":{"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":2}]},"1":{"r":[{"g":2,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":1,"b":[],"s":1},{"0":{"r":[{"g":1,"v":0}]},"1":{"r":[{"g":1,"v":1}]},"t":1,"b":[],"s":1},{"0":{"r":[{"g":3,"v":0}]},"1":{"r":[{"g":3,"v":1}]},"t":1,"b":[],"s":0},{"0":{"r":[{"g":15,"v":0}]},"1":{"r":[{"g":15,"v":1}]},"t":1,"b":[],"s":0}]}

RavenSystem, 2019-12-12:
Try v0.8.3.

skv6, 2019-12-12:
In 0.8.3 version I added 8CH to HomeKit. In my JSON version its lost signal and doesn't work after, but it can be added. I made this JSON at haa-configurator.

RavenSystem, 2019-12-12:
Try using gpio 16 instead gpio 2.

skv6, 2019-12-13:
Use gpio 16 instead gpio 2. its can be added to HomeLit but still doesht work there. 1 times it worked 30 sec and no answer after. And cant enter to setup mode (powered cut or freezes before 3 seconds since boots).

{"c":{"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":16}]},"1":{"r":[{"g":16,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":1,"b":[],"s":1},{"0":{"r":[{"g":1,"v":0}]},"1":{"r":[{"g":1,"v":1}]},"t":1,"b":[],"s":1},{"0":{"r":[{"g":3,"v":0}]},"1":{"r":[{"g":3,"v":1}]},"t":1,"b":[],"s":0},{"0":{"r":[{"g":15,"v":0}]},"1":{"r":[{"g":15,"v":1}]},"t":1,"b":[],"s":0}]}

RavenSystem, 2019-12-17:
Logs please.

skv6, 2019-12-19:
[screenlog.0.zip](https://github.com/RavenSystem/esp-homekit-devices/files/3983769/screenlog.0.zip)

RavenSystem, 2019-12-19:
There is nothing in log. In order to see UART output, you must include "o":1 in your JSON config.

skv6, 2019-12-19:
{"c":{"b":[{"g":0,"t":5}],"o":1},"a":[{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":16}]},"1":{"r":[{"g":16,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1}]},"b":[],"s":1},{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":1,"b":[],"s":1},{"0":{"r":[{"g":1,"v":0}]},"1":{"r":[{"g":1,"v":1}]},"t":1,"b":[],"s":1},{"0":{"r":[{"g":3,"v":0}]},"1":{"r":[{"g":3,"v":1}]},"t":1,"b":[],"s":0},{"0":{"r":[{"g":15,"v":0}]},"1":{"r":[{"g":15,"v":1}]},"t":1,"b":[],"s":0}]}

[screenlog.0.zip](https://github.com/RavenSystem/esp-homekit-devices/files/3984586/screenlog.0.zip)

RavenSystem, 2019-12-19:
The log only shows that device is in setup mode scanning wifi networks.

skv6, 2019-12-19:
After firmware I can add device to HomeKit and it works 30 seconds. After that it doesnt work, HomeKit cant see it and after restart still doesnt work.

RavenSystem, 2019-12-19:
Logs are necessary to understand what is happening.

closed 2020-01-10 by RavenSystem