RavenSystem/esp-homekit-devices #658: HAA - HomeKit App reject QR Code as it is already taken

Shimakuchi, 2019-12-12:
I flashed a shelly 2.5 with the older RavenCore Firmware (need the capability of control Roller Shutter). As described here the code gets rejected.

Now I flashed a shelly 1 with the same result.

[image]

_Originally posted by @Shimakuchi in https://github.com/RavenSystem/esp-homekit-devices/issues/657#issuecomment-565039540_

PandaBoba, 2019-12-12:
I have the same problem with newest HAA

mafyata, 2019-12-12:
Don’t scan , type the code manually 02182017

PandaBoba, 2019-12-12:
Then in my case it doesn't search for any accessory

mafyata, 2019-12-12:
Did you click on this button ?
[image]

PandaBoba, 2019-12-12:
[image]
Yes

PandaBoba, 2019-12-12:
Ok for me it was wrong json. Now everything working fine

Shimakuchi, 2019-12-12:
Hi @mafyata. Yes I did try to enter the code manually

@PandaBoba In case of my shelly 1 (HAA) the faulty json could be the problem. I thought I could add it to HomeKit and then configure the json.

In case of the shelly 2.5 (RavenCore) there is no json. The device can be configured after adding it to HomeKit via the Eve app

RavenSystem, 2019-12-17:
RavenCore has not a QR-Code. You must use simple code `021-82-017`.

HAA has QR-Code, but you must setup JSON first.

closed 2019-12-30 by Shimakuchi

pilot1981, 2019-12-31:
I notice some different behaviour using QR code and normal code: sometime when QR code don't work I have to use normal code...