RavenSystem/esp-homekit-devices #660: Shelly1 - GarageDoor JSON without sensors

fIctioN600, 2019-12-13:
Anyone knows how the JSON have to look like?

RavenSystem, 2019-12-13:
Don’t use f2, f3, f4 or f5. 

For example, Sonoff with HomeKit Garage Door adaptation (Working time: 18 seconds. With stop-step. Dry relay connected to action garage door button. No sensors):

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2}}]}

jakudo, 2019-12-15:
Hello, currently I have Sonoff 5V with ravencore. I'd like to update it to HAA (since it supports Garage door). I also have it without sensor. I have a few questions to understand what it does:

- when I click on the icon in Homekit it starts to open the door
- after 18 sec it shows door "open"
- when I click on the icon again, it starts closing the door
- after 18 sec it shows door "closed"
- what happens when the door is opening/closing and I press the button in Homekit again (within the 18 secs interval)? I guess it stops opening/closing

Is that correct?

vsidlu, 2019-12-27:
@RavenSystem please help with json for -
no senors. gate opens in 18 sec and automaticlly closes after 30 sec without button push.

RavenSystem, 2020-01-10:
@vsidlu Not possible with HAA.

closed 2020-01-10 by RavenSystem

vsidlu, 2020-01-10:
found workaround - 
{"c":{"o":1,"l":2,"i":1,"b":[{"g":0,"p":1,"t":1}]},"a":[{"t":40,"d":5,"0":{"r":[{"g":12,"i":0.5}]},"1":{"a":0},"2":{"a":0},"3":{"a":2}},{"t":1,"i":25,"0":{"r":[{"g":14}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":14,"p":1,"t":1},{"g":14}]}]}

made dummy button which return off in 25 sec after turend on. 
then made home automation
- when button off gate closes
- when button on gate opens
- when gate opens button on

so after I push to open gate button is turend on and after 25 sec it turns off and if it is off gate is closing

RavenSystem, 2020-01-10:
Yes, that works, but why do you not use sensors? With sensors your door will be smarter and you will not need any workaround.

vsidlu, 2020-01-10:
because it is very easy. i open gate remote and add to sonoff relay.