RavenSystem/esp-homekit-devices #663: External toggle problem

tmps99308091, 2019-12-14:
The toggle now will turn on the switch when pressed and turn off when released. How can I set up an external toggle which will turn on the switch when I press and release a button, and then turn it off when I press and release it again. The json string now is `{"c":{"o":1},"a":[{"0":{"r":[{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1}]},"t":1,"b":[],"s":0,"f0":[{"g":16,"t":1,"p":1}],"f1":[{"g":16,"t":0}]},{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":1,"b":[],"s":0,"f0":[{"g":4,"t":1,"p":1}],"f1":[{"g":4,"t":0}]}]}`

diodo157, 2019-12-14:
You must change the part:
"b": [],
with (putting the right gpio)
"b": [{"g": 0, "t": 1}],

tmps99308091, 2019-12-14:
Thanks for the help. Should I remove the string inside “f0” and “f1”

RavenSystem, 2019-12-19:
@tmps99308091 yes.

closed 2019-12-19 by RavenSystem