RavenSystem/esp-homekit-devices #664: Some extra settings for thermostat

diodo157, 2019-12-14:
I have three thermostats in the house, with ds18b20, but they are too sensitive, so nothing is enough that the heating part, having a pellet boiler, creates some problems.
It would be interesting to put parameter to make an average of "x" readings to get more reliable reading.
Thanks

Bolshoylu, 2019-12-16:
May be you can use "d": temperature deadband to adjust how thermostat works. It help for me :)

GPL71, 2019-12-17:
What is deadband? Is it the range where thermostat in inactive? If so, how does it work?

RavenSystem, 2019-12-17:
Example of deadband and heater:

You have a heater with a deadband of 5º that heats until 30º.
Heater will heat and it stops when 30º is reached.
And it will heat again when temperature is less than 25º (30º - deadband).

Bolshoylu, 2019-12-18:
I use d = 0.5 and if the thermostat is set to a temperature of 30 degrees, then it heats up to 30.5 and does not turn on to 29.5.  the relay spoils so less and does not click often

diodo157, 2019-12-18:
Hi,
Thanks Bolshoylu, I already use it.
From the tests that I have carried out the value is only in negative, that is when it drops to 29.5 it turns on until it goes off when it reaches 30, we call it hysteresis.
But the 18b20 sensors are too fast, and it is enough that it passes close to change the reading with the movement of the air, and the heating starts, and then goes out after a very short time.
Pellet boilers are not like gas boilers, they have a lighting ignition procedure that takes time and consumes unnecessarily.

diodo157, 2019-12-18:
Now I tried to put the waterproof sensor on a thermostat, and with the coating it is less fast and seems to work better

closed 2020-01-10 by RavenSystem