RavenSystem/esp-homekit-devices #667: Freeze sono 4ch

simoneorsi, 2019-12-14:
I upgrade my son off 4ch pro to the last one version.
If I switch fast more and more time the 4 switch the system freeze.
I tried to take a log file but my Mac terminal windows leave blank without nothing message.
thanks

skv6, 2019-12-15:
Sonoff mini too. Work some seconds after that freeze. 0.8.6

hillaliy, 2019-12-17:
Sonoff T1 US 2 Gang freeze too. 0.8.6

RavenSystem, 2019-12-17:
JSON string and logs, please.

hillaliy, 2019-12-18:
How i get logs?
JSON: {"c": {"l": 13,"b": [{"g": 0,"t": 5}],"i": 1},"a": [{"0": {"r": [{"g": 12}]},"1": {"r": [{"g": 12,"v": 1}]},"b": [{"g": 0}],"s": 0},{"0": {"r": [{"g": 5}]},"1": {"r": [{"g": 5,"v": 1}]},"b": [{"g": 9}],"s": 0}]}

RavenSystem, 2019-12-18:
I tested your JSON with a protoboard and there is not any freeze.

You can see how capture logs here in the Wiki:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-Instructions#capture-logs

simoneorsi, 2019-12-19:
I try to do how explain in to the instruction to capture log but the window dindn’t send any info. To calture log I need to use the same rs232 usb converter that I use to program the sonoff? Only rx and tx or another wire. Sorry for my incompetence!

RavenSystem, 2019-12-19:
Sorry, yes. You must connect your device only to your rs232 (rx, yx, vcc and ground). Don’t connect it to 230VAC.

hillaliy, 2019-12-19:
Sorry it dosn't freeze it get to setup mode. I add Setup Mode Time to Json hope it will help.

RavenSystem, 2019-12-19:
If device enters in setup, is because you toggle quickly switches up to 8 times in a row. You can use setup mode time key or new "z" key. Check this: https://github.com/RavenSystem/esp-homekit-devices/releases

closed 2019-12-19 by RavenSystem