RavenSystem/esp-homekit-devices #668: MagicHome RGB LED Strip Controller doesn't work with all colors correctly

x3ngeri, 2019-12-15:
Hey, first of all thanks for this great project! i've used one of the first "RavenCore" version 0.7.3 on my Sonoff-Basic, it works still today very great.

Now i've bought some MagicHome RGB LED Strip Controller, but after programming with HAA 0.8.6 the colors doen't match to the pins. Always one color is missing. And yes its an RGB-controller-version...

Is somebody who knows informations about it? I used the JSON: 

> MagicHome RGB LED Strip Controller (R pin 15, G pin 13, B pin 12):

{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":15,"g":13,"v":12,"b":[{"g":0}]}]}

mafyata, 2019-12-16:
Try other gpio.
Mine is: "r":5,"g":12,"v":13

x3ngeri, 2019-12-16:
> Try other gpio.
> Mine is: "r":5,"g":12,"v":13

Thx, i'll check it and give feedback.

x3ngeri, 2019-12-16:
> Try other gpio.
> Mine is: "r":5,"g":12,"v":13

Hey, thanks... thats was the right JSON. Now it works perfectly, so i can order more of these controllers.

mafyata, 2019-12-16:
If it is not enough brighter you can try to add this in json:
"fr":5,"fg":5,"fv":5
Mine one work with this , other not. You can test.

closed 2019-12-17 by RavenSystem