RavenSystem/esp-homekit-devices #669: Auto OTA on Boot?

noobydp, 2019-12-16:
Hi Jose, I was wondering if you’d consider letting the device auto-update via OTA when its rebooted, if the Auto OTA Update option was selected when first setup?

I’ve setup a few devices and I want to keep them up to date, but they’re in inconvenient places so pressing the setup button when it powers on will be difficult.

Thanks

St0nehead, 2019-12-16:
Agree!

sancheztineo, 2019-12-16:
I add a dummy switch with JSon. When I want update, with app home, I press 8 times the dummy switch and the devices updates the firmware.

RavenSystem, 2019-12-17:
Full auto updates is not a good idea, because boot will be very slow, and there is a risk that device enters into an infinite loop.

@sancheztineo that is a very good idea. And you can put those dummy switches at other room.

noobydp, 2019-12-17:
Thanks!
The dummy switch is a great idea.

closed 2019-12-17 by noobydp

sancheztineo, 2019-12-18:
I’ve put a room with all Dummy Switch for OTA Updates. 
[image]

mafyata, 2019-12-18:
I have one idea. Can you make option in HAA , to enable setup button instead of dummy switch? When I press setup button to enter setup mode. 1 press instead of 8 press.

RavenSystem, 2019-12-19:
This is better: https://github.com/RavenSystem/esp-homekit-devices/releases/tag/HAA_0.8.7