RavenSystem/esp-homekit-devices #672: Shelly 1 - Garage Door with external push button [HAA]

maciekfreak, 2019-12-18:
I have a problem with configuration of HAA into shelly 1 with Garage door and external push button to open/stop/close doors.

My actual configuration:

{"c":{"b":[{"g":5,"p":0,"i":1,"t":5}]},"a":[{"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":4,"i":1.5},{"g":4,"v":1,"i":0.5},{"g":4,"v":1,"i":2}]},"3":{"a":2},"t":40,"d":22}]}

Using homekit garage door are opening and closing correctly. But when using external button nothing happened.

RavenSystem, 2019-12-18:
That is because you have not configured any external button to manage door (`"b":[{"g":5}]` is missed). Try this:

`{"c":{"b":[{"g":5,"p":0,"i":1,"t":5}]},"a":[{"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":4,"i":1.5},{"g":4,"v":1,"i":0.5},{"g":4,"v":1,"i":2}]},"3":{"a":2},"t":40,"d":22,"b":[{"g":5}]}]}`

maciekfreak, 2019-12-18:
I had the **{"b":[{"g":5,"p":0,"i":1,"t":5}]}** at the beginning. Is it the same?

diodo157, 2019-12-19:
No, this is needed to be able to go into setting mode through the long press on g5, and not to execute commands

closed 2020-01-10 by RavenSystem

aleroot, 2020-10-13:
I have tried this configuration and it works from homekit, I cannot make it work the push button, since at the moment the push button poles are connected the same way Shelly I and O are, so basically when the button close the two contact the Garage Door open or close, the same way the Shelly open or close. 

How should I connect the push button in this case ? 

Thanks