RavenSystem/esp-homekit-devices #674: Help with restAPI feature

peros550, 2019-12-19:
Would it please be possible to provide some description how we could use the new restAPI feature?

I have a sonoff basic and would like to see how I could control it/check status using restAPI. 

Thanks & Regards

RavenSystem, 2019-12-19:
https://github.com/RavenSystem/esp-homekit-devices/wiki/General-Configuration#enable-homekit-rest-api

https://twitter.com/RavenSystem/status/1200011567011684352

peros550, 2019-12-21:
Thank you man! I found my way! 

Let me ask something. In case the device change status (i.e. through a command received by Apple TV, an inching function, etc) is it possible to have an http call to a server to report the new status? Or the server is supposed to poll the device to get the status?

And another question, when getting the status, is it possible to include Wifi signal status, and any other information related to wifi connection , like connection uptime etc. 

Thanks !

RavenSystem, 2019-12-21:
> Let me ask something. In case the device change status (i.e. through a command received by Apple TV, an inching function, etc) is it possible to have an http call to a server to report the new status? Or the server is supposed to poll the device to get the status?

Actually server must get device status. In the future, you will can setup a HTTP get Action to avoid server polls.

> And another question, when getting the status, is it possible to include Wifi signal status, and any other information related to wifi connection , like connection uptime etc.

Not, only HomeKit information.

peros550, 2019-12-21:
Thank you Jose!

closed 2019-12-21 by peros550

i3laze, 2020-01-22:
> Would it please be possible to provide some description how we could use the new restAPI feature?
> 
> I have a sonoff basic and would like to see how I could control it/check status using restAPI.
> 
> Thanks & Regards

In case someone else looking for API examples, here’s a script I use to toggle HAA remotely via it’s API:

#!/bin/bash
IP=192.168.1.129
aid=1
iid=9

result=$( curl -s http://$IP:5556/characteristics?id=$aid.$iid | jq -r '.characteristics[].value' )

case $result in
  true)
    echo "Toggling off.."
    curl -X PUT -d '{"characteristics":[{"aid":'$aid',"iid":'$iid',"value":false}]}' http://$IP:5556 
    ;;
  false)
    echo "Toggling on.."
    curl -X PUT -d '{"characteristics":[{"aid":'$aid',"iid":'$iid',"value":true}]}' http://$IP:5556 
    ;;
  *)
    echo "Invalid cURL response!"
    exit 1 ;;
esac

i3laze, 2020-04-18:
Just noticed that _jq_ above isn't really nesessary (as it is not available for arm64). case matches wildcard *substring* just fine:
```
#!/bin/sh
IP=10.0.0.92
aid=1
iid=9

result=$( curl -s http://$IP:5556/characteristics?id=$aid.$iid ) 
case $result in
  *true*)
    echo "Toggling off.."
    curl -X PUT -d '{"characteristics":[{"aid":'$aid',"iid":'$iid',"value":false}]}' http://$IP:5556
    ;;
  *false*)
    echo "Toggling on.."
    curl -X PUT -d '{"characteristics":[{"aid":'$aid',"iid":'$iid',"value":true}]}' http://$IP:5556
    ;;
  *)
    echo "Invalid cURL response!"
    exit 1 ;;
esac
```