RavenSystem/esp-homekit-devices #675: Setup new feature - system actions

mafyata, 2019-12-19:
Hi , how to setup this new feature - system actions?

RavenSystem, 2019-12-21:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#system-actions

homkittoby, 2019-12-21:
Idea!
Is the device in setup mode and nothing happens for 2 minutes. The device boots again in normal mode.
So the device does not get caught in setup mode if it is started accidentally.

mafyata, 2019-12-21:
{"c":{"n":"Hol-Led","b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"1":{"s": [{ "a": 1}]},"r":5,"g":12,"v":13,"fr":5,"fg":5,"fv":5}]}
What is wrong?

RavenSystem, 2019-12-21:
What is the issue?

mafyata, 2019-12-21:
The led strip is off. Press 1 time turn the light on and the second press enter setup mode instead of turning off the light.

RavenSystem, 2019-12-21:
Well, order to enter setup mode is coming with your first press: "1":{"s": [{ "a": 1}]}, but there is a delay of 3 seconds until device reboots.

mafyata, 2019-12-21:
But it is with mine main button, not second for setup mode. I think that with this function I will receive second button for this device for entering setup mode.

RavenSystem, 2019-12-21:
Your setup has one accessory, and you set "1" to enter setup mode. Action "1" is executed when light is turned on.

mafyata, 2019-12-21:
Hmmm. Can you help me?
This is my json: 
{"c":{"n":"Hol-Led","b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13,"fr":5,"fg":5,"fv":5}]}
Can you tell me the correct json with second button for entering setup mode?

homkittoby, 2019-12-21:
{"c":{"n":"Hol-Led","b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13,"fr":5,"fg":5,"fv":5},{"0":{"r":[]},"1":{"r":[],"s":[{"a":1}]},"t":1,"b":[],"s":0}]}

???

homkittoby, 2019-12-21:
@RavenSystem 

**Thank you** 

(Changelog: General: Now device in setup mode will reboot in normal mode automatically after 90 seconds is setup web page is not loaded and Auto OTA Updates is disabled.)

closed 2019-12-21 by RavenSystem

mafyata, 2019-12-21:
It works. Thanks.

mafyata, 2019-12-21:
why it is not working on my other device? i copy the setup you added to my 1st device.
{"c":{"n":"Kuhnq-Led","b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13},{"0":{"r":[]},"1":{"r":[],"s":[{"a":1}]},"t":1,"b":[],"s":0}]}

RavenSystem, 2019-12-21:
Try reset HomeKit ID.

mafyata, 2019-12-21:
I forgot to update from 0.8.2 Not it is ok.