RavenSystem/esp-homekit-devices #676: Floorheating: Combining thermometer with sonoff basic for ex.

URBANsUNITED, 2019-12-19:
Hi!

First of all: Thanks for this great project! Brilliant!! Thanks

I have a feature request.
Do you see any possibility for combining two devices into one?
Currently I am using for my floor heating each a sonoff basic flashed with HAA (for opening and closing the valve) and a thermometer from Xiaomi (TemperatureAndHumiditySensor for sending the temperature) 
I would like to make a thermostat out of these two devices.
A different setup is not really possible. The sonoff is next to my heating system and a thermometer in each room. Both absolutely hidden. Adding a thermometer to the basic makes no sense as it is not installed in each room.
I have the UUID for each thermometer maybe that's the catch?

Many thanks in advance!!

Cheers

URBANsUNITED

diodo157, 2019-12-20:
I don't know if I understood your problem.
If you want to use your devices as a thermostat, you can use the EVE program, which allows you to create rules based on the temperature read by the sensor and turn the sonoff on and off

URBANsUNITED, 2020-01-07:
Hi!

I am really sorry to response that late. Sorry!

The EVE route is my current setup I am using for 2 years already:
Ex.
If the temperature is any changing and is < 22,5°, turn on sonoff basic
If the temperature is any changing and is > 22,5°, turn off sonoff basic

My personal problem is that I want this in one device like a thermostat. I want to get rid from two devices in my dashboard. I want one device.
Like an external temperature sensor attached to the sonoff basic.
But my temperature sensor is wireless and hidden, the  DHT22 needs to be get wired to the sonoff which makes it complicated to use as a thermostat
My sonoffs are all inside the floorheating wall panel and short way connected with the valves to open/close the heating

Maybe I want too much, but it looks much nicer… and would be a very cool feature!

Or a different way would be to classify a device to be a thermostat which is a sonoff device somehow…

The problem is that the floorheating is very limited with homekit devices. You have a thermometer like from Nest and that’s it. There must be a cheap and easy plus fancy way…

Maybe you see any way to make this possible.

Many thanks!!!

Best regards

URBANsUNITED

RavenSystem, 2020-01-10:
Wireless sensors are not supported, and there are no plans to support it, sorry.

closed 2020-01-10 by RavenSystem