RavenSystem/esp-homekit-devices #677: Thermostat problem, no change state to active

crazyclimber999, 2019-12-21:
Hi jose, Gianluca help me to make json for thermostat with son off th16.

However it doesn’t work properly, the state is always inactive. This is the json:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"a":0},"3":{"r":[{"g":12,"v":1}]},"5":{"a":0},"t":21,"s":5,"g":14,"b":[],"n":2,"z":0.5,"m":5,"x":25,"d":26}]}

I try to enter in config mode pressing 8 times the button but I can’t..

Please help me.

diodo157, 2019-12-21:
With this setup you can enter setup mode by holding the button down for 10 seconds

crazyclimber999, 2019-12-21:
thank you, and why it don't change the state to active?

RavenSystem, 2019-12-21:
You use a dead band of 26º, that is crazy. If you want to use this feature, better use a value between 0 and 1 (you can use decimals). For example `"d":0.5`

And if you want to turn on/off thermostat with Sonoff built-in button, you must add `{"g":0}` into thermostat empty array of digital inputs `b:[]`.

crazyclimber999, 2019-12-24:
Thank you… I fix it!

Andrea Berti
Mob. (+39) 335252416
http://www.andreaberti.it <http://www.andreaberti.it/>

closed 2019-12-24 by RavenSystem