RavenSystem/esp-homekit-devices #679: Rocker switch as stateless button single press

danielhelmstedt, 2019-12-21:
Hi, is it possible to set up a standard rocker switch/lightswitch to act as a single button press when toggled?
 I tried setting 2 single press state inputs - one default and one falling edge, both GPIO12 for my sonoff mini, but no luck. Do I need to invert one or something?

danielhelmstedt, 2019-12-21:
Just realised GPIO 12 is the relay and I want GPIO 4, duh 🙄

closed 2019-12-21 by danielhelmstedt

rasod, 2020-02-22:
@danielhelmstedt i'm trying something similar what configuration and wiring did you end up using?

danielhelmstedt, 2020-02-23:
Just wire the switch up normally to a GPIO, and configure 2 single press inputs - one with default activation and the other with falling edge. Then you’ll probably want to configure the single press button in HomeKit to use a shortcut with an If statement (I.e. if light is ON, turn OFF, else turn ON)

chenfengfeng, 2022-02-19:
@danielhelmstedt Can you share your JSON configuration?