RavenSystem/esp-homekit-devices #680: Bootloop after OTA installation

peros550, 2019-12-21:
Here is the following log. It repeats over and over. Right after OTA installation finishes and HAA takes control I get the following: 

```
connected with Test1, channel 6
dhcp client start...
ip:192.168.177.154,mask:255.255.255.0,gw:192.168.177.1
HAA > Disarming Emergency Setup Mode
rm match
del if0
usl
sul 0 0

ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x40100000, len 2292, room 16

0x40100000: _stext at ??:?

tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeffc0, prio:14, stack:512
pm_task_hdl : 3ffef8b0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3fe0, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Dec 20 2019 23:02:12
phy ver: 273, pp ver: 8.3

HAA ! Invalid JSON
mode : sta(cc:50:e3:57:4f:dc)
add if0
scandone
add 0
aid 12
cnt

connected with Test1, channel 6
dhcp client start...
ip:192.168.177.154,mask:255.255.255.0,gw:192.168.177.1
HAA > Disarming Emergency Setup Mode
rm match
del if0
usl
sul 0 0

ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x40100000, len 2292, room 16

0x40100000: _stext at ??:?

tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeffc0, prio:14, stack:512
pm_task_hdl : 3ffef8b0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3fe0, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Dec 20 2019 23:02:12
phy ver: 273, pp ver: 8.3

HAA ! Invalid JSON
mode : sta(cc:50:e3:57:4f:dc)
add if0
scandone
add 0
aid 12
cnt

connected with Test1, channel 6
dhcp client start...
ip:192.168.177.154,mask:255.255.255.0,gw:192.168.177.1
HAA > Disarming Emergency Setup Mode
rm match
del if0
usl
sul 0 0

ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x40100000, len 2292, room 16

0x40100000: _stext at ??:?

tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeffc0, prio:14, stack:512
pm_task_hdl : 3ffef8b0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3fe0, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Dec 20 2019 23:02:12
phy ver: 273, pp ver: 8.3

HAA ! Invalid JSON
mode : sta(cc:50:e3:57:4f:dc)
add if0
scandone
add 0
aid 12
cnt

```

The device is a Sonoff S20. I have erased flash and re-installed firmware many times. I have tried to install other custom firmware (not OTA method) and it works fine. Whenever I try to install HAA using OTA I get this strange boot loop. 

The only thing I might did wrong, was that at my first try to install HAA, at the LCM page, I successfully chose the wifi network but did not set the Ravensystem/haa repo. 

Could that create a problem? I tried to erase_flash since that try and the problem persists. 
Any ideas?

pilot1981, 2019-12-21:
Who is richardaburton@gmail.com?
@RavenSystem you use his boot loader?

peros550, 2019-12-21:
> Who is [richardaburton@gmail.com](mailto:richardaburton@gmail.com)?
> @RavenSystem you use his boot loader?
@pilot1981 here is his repo:
https://github.com/raburton/rboot

@RavenSystem , thanks for looking at this.

RavenSystem, 2019-12-21:
@peros550 It is a bug introduced with last version. It is fixed now in v0.8.10.

@pilot1981 I use boot loader from esp-open-rtos.

RavenSystem, 2019-12-21:
This bug does not affect to updated devices with a valid JSON.

@peros Many thanks for feedback and logs. They let me fix it in a few minutes.

closed 2019-12-21 by RavenSystem