RavenSystem/esp-homekit-devices #681: Problem to upgrade to RC 1.3 using OTA

pilot1981, 2019-12-22:
Hi,
I have sonoff with RAVENCORE 1.1, but it I try to upgrade to 1.3 using Eve App nothing happen...
What is wrong?

Regards

pilot1981, 2019-12-22:
I tried again: sonoff died! No AP MODE...
Suggestions?

mafyata, 2019-12-22:
Try to flash again.

pilot1981, 2019-12-22:
Very confortable...

GPL71, 2019-12-22:
Hey guys... if I understand well, RavenCore 1.3 is only an automatic procedure for upgrading to HAA.
Then it’s necessary have a valid json to complete operation.

@RavenSystem Josè help is necessary

mafyata, 2019-12-22:
No , you are wrong. 1.3 is security update.

RavenSystem, 2019-12-22:
@GPL71 you are right. 

@mafyata you are wrong. I said that RaveCore will not be updated anymore. And now HAA supports all RC features. Please, read release notes and Wiki information about that.

pilot1981, 2019-12-22:
ok, now I have some problem with JSON of garage door, I try using discord...but with a shelly 1 I receive error message: this device is already added to homekit...
how I can solve it?

I already removed it from my home app...

GPL71, 2019-12-23:
@pilot1981 Did you flag HomeKit Reset ID?

pilot1981, 2019-12-23:
no, I have to do it? I already removed device from my home app..

GPL71, 2019-12-23:
yes ... try it too ..

pilot1981, 2019-12-23:
I tried many times to reset HomeKit, but this don’t solve my problem using QR CODE.

Only solution is to use normal code...

closed 2019-12-23 by pilot1981