RavenSystem/esp-homekit-devices #682: Shelly 1PM not working

ArFurik, 2019-12-22:
Hi, and thank you for this amazing work.
I flashed Shellys 1 without problem and they work fine. (main 0.9.1)
Now I tried to flash a Shelly 1PM with the same firmware, cause it seems very similar on commands and GPIOs, even if the power port is a little bit different due at the ability to measure the consummation of the device.
But, if all is well flashed and recognised by the Apple Home application, the switch does not work and doesn't comute.
Is there a specificity for this device ? or a different JSON to put in ?
Has somebody already well install this Shelly 1PM device ?
Thank you for your help.

GPL71, 2019-12-23:
Have you GPIO for shelly 1PM?

ArFurik, 2019-12-23:
The GPIO basic map is the same than shelly one's, as you can see below:
[image]
The power port is not quite the same of the Shelly 1, but I'm sure that is could only be the way to connect the output relay to be able to measure the power consummation.
Of course, this measure and a led status could use other GPIO pins but I don't care about for the moment. This Shelly model is useful for me due at the overheat protection included.
See the cabling schematic below:
[image]

RavenSystem, 2019-12-23:
It uses same Shelly1 GPIOs for relay and switch, I don't understand why it doesn't work.

I am not sure if overheat protection will work. If it is a hardware feature, then yes, but if it is software based, no.

ArFurik, 2019-12-23:
It's also puzzled me. I ask about that for this reason, if someone has already flashed a 1PM model this can help me to know if a special feature JSON is necessary, or if my device is simply not good and not working.

RavenSystem, 2019-12-23:
@ArFurik Can you test if external switch works?

RavenSystem, 2019-12-23:
Another question, are you eating relay with device powered with USB-adapter or 230VAC?

ArFurik, 2019-12-23:
It doesn't work. On the Apple Home app, it seems to work perfectly, the response at a command from my iPhone is good, but in fact, the light connected to the switch is not working. No typical "click" can be eared indicating the switch is commutating, and the real switch button connected to the Shelly doesn't change the state, nor on the Home App.
So, in fact, the soft and wifi seem to work, but the power is not commutated and the real switch is not acting on the state.

I unplug all the wires connected to the USB adapter when I test with the power AC. I never usethe both in the same time.

RavenSystem, 2019-12-23:
Try this JSON. I got GPIO info from Tasmota template (https://templates.blakadder.com/shelly_1PM.html):

`{"c":{"l":0},"a":[{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":4,"p":0},{"g":4,"t":0}]}]}`

ArFurik, 2019-12-23:
Gracias José, I'll try it today and tell you if it changes the behavior.

RavenSystem, 2019-12-23:
Tasmota template uses:
- GPIO-0 -> Inverted status LED.
- GPIO-4 -> External Switch, no pullup resistor.
- GPIO15 -> Relay.

It is totally different from official info.

RavenSystem, 2019-12-23:
Other thing. In Tasmota template, there is nothing configured as temperature sensor to avoid overheat, so I think it is a hardware protection, unless overhead is based on power meter, but in that case is not overheat protection, it is overpower protection. Shelly 2.5 has a temperature sensor to do overheat protection, but it is software based.

ArFurik, 2019-12-25:
¡ Holá José, y feliz Navidad ! 
I've just tried your JSON from Tasmota template... **It works !**
Thank you very much for your great work and your responses. Like you, I think that the overheat protection is a hardware protection (it's described as it in the specs).
It's now possible to flash shellys 1PM.

closed 2019-12-25 by ArFurik

pilot1981, 2019-12-25:
You also have power meter feature?

ArFurik, 2019-12-25:
> You also have power meter feature?

No, I think it'll be possible to catch it knowing the concerned GPIO. Try to adjust an accurate JSON and tell us if you find the solution.