RavenSystem/esp-homekit-devices #684: problem about JSON config garage door device type on HAA

pilot1981, 2019-12-23:
Hi,
I have some problem to migrate from RC to HAA for my garage door. My garage door work in this way:
- one impulse on button to open garage
- one impulse on button to close garage
- garage automation have already obstructions sensor for security matter
I also use physical button on sonoff to control garage door when I don't have smartphone with me. 
I'm using Sonoff basic R2 with cutted track. I attached close sensor to GPIO14.
I started from JSON example from wiki:

Sonoff with HomeKit Garage Door adaptation (Working time: 18 seconds. Only close sensor connected to GPIO 14 and stop-step. Dry relay connected to action garage door button):

* {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":14,"t":0}],"f4":[{"g":14}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2}}]}

I would understand better if this JSON is correct for my configuration:
- you confirm GPIO14 is correct for close sensor? It seem Sonoff don't feel closed status
- can you give me a JSON expample without close sensor to do some test on my Sonoff?
- with this problem on close sensor, I have always garage door in closing state; why relay don't start? 
- external button of sonoff is already enabled in this JSON? how I can do it?
- to enter in setup mode you suggest to create a dummy button, you can give me correct JSON?
- which is initial state?

RavenSystem, 2019-12-23:
> you confirm GPIO14 is correct for close sensor? It seem Sonoff don't feel closed status

Yes,`"f3"` and `"f4"`, but maybe your sensor works in the opposite way:
Try replacing `"f3":[{"g":14,"t":0}],"f4":[{"g":14}]`with `"f3":[{"g":14}],"f4":[{"g":14,"t":0}]`

Complete JSON:
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":14}],"f4":[{"g":14,"t":0}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2}}]}`

> can you give me a JSON expample without close sensor to do some test on my Sonoff?

Only remove `"f[2..7]"`:
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2}}]}`

> with this problem on close sensor, I have always garage door in closing state; why relay don't start?

Maybe your sensor works in the opposite way.

> external button of sonoff is already enabled in this JSON? how I can do it?

Like any other accessory, adding `"b"` array to the accessory root. Using GPIO-0 button: `"b":[{"g":0}]`

> to enter in setup mode you suggest to create a dummy button, you can give me correct JSON?

`{"1":{"s":[{"a":1}]}}`

> which is initial state?

Initial state is based on sensors. In your case, based on sensor close: if sensor detect door is closed, then initial state is closed, if not, it is opened.

This is JSON with inverted sensor, built-in button to operate door, and HomeKit switch to enter setup mode:
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":14}],"f4":[{"g":14,"t":0}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"b":[{"g":0}]},{"1":{"s":[{"a":1}]}}]}`

You can find all info about garage door here:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Garage-Door

pilot1981, 2019-12-23:
sorry but last JSON create two switches (no garage type accessory?): I can't see close sensor status...
If I press both switches button one times, device don't answer again in homekit...I try to go in setup mode pressing 8 times external button but nothing...so I flashed again sonoff...
any idea what happened?

RavenSystem, 2019-12-23:
You must reset HomeKit id when JSON changes number of accessories or HomeKit functions.

pilot1981, 2019-12-23:
I tried also the first JSON:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":14}],"f4":[{"g":14,"t":0}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2}}]}

but I see a garage accessory always on opening state (don't read close sensor) and a switch (dummy?):
-dummy switch some time is unavailable
-garage door some time is unavailable

I also tried a new hardware of Sonoff basic R2 with cutted track with same connection...but it don't solve...

I'm very sad...I loved RavenCore...

where am I wrong?

pilot1981, 2019-12-24:
yes, I reset Homekit ID every time I try a new JSON...I don't understood something...

pilot1981, 2019-12-25:
ok, I done some test with new hardware, always sonoff basic R2 with cute track (best solution for me: power supply included).

this result of my tests:

>>>>>>>OFFICIAL EXAMPLE:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":14,"t":0}],"f4":[{"g":14}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2}}]}
->>>>>garage door remain always in opening status, closed sensor don't work:  it's not possible to use it because relay don't start never

>>>>>>FIRST @RavenSystem SUGGESTION WITH CLOSE SENSOR IN OPPOSITE WAY:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":14}],"f4":[{"g":14,"t":0}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2}}]}
->>>>>>garage door remain always in opening status, closed sensor don't work:  it's not possible to use it because relay don't start never

>>>>>SECOND @RavenSystem SUGGESTION WITHOUT SENSORS:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2}}]}
->>>the work fine, but I can it use in production environment because it isn't not secure to know the real status of door...In this config I don't have physical button working...

>>>>>THIRD @RavenSystem  SUGGESTION with inverted sensor, built-in button to operate door, and HomeKit switch to enter setup mode:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":14}],"f4":[{"g":14,"t":0}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"b":[{"g":0}]},{"1":{"s":[{"a":1}]}}]}
->>>>garage door remain always in opening status, closed sensor don't work: it's not possible to use it because relay don't start never

So my major problem seems on close sensor...

So, requirements are these:
- use sonoff basic R2 with cutted track
- device type 40 for garage door
- use close sensor on GPIO14
- use physical button
- have dummy switch to enter in setup mode

In all my tests, I have problem to return in setup mode, so for every test I have to flash again...I use external button or dummy switch for 8 times: it's right? where is the trick?

any idea?

P.S. for now no any sonoff bricked...but sonoff have low quality so...

pilot1981, 2019-12-25:
using JSON template in your repository I found these logs:

>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: Got new client connection: 5 from 192.168.1.113
>>> HomeKit: Got new client connection: 6 from 192.168.1.157
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> HomeKit: [Client 6] Pair Verify Step 1/2
>>> HomeKit: [Client 4] Pair Verify Step 2/2
>>> HomeKit: [Client 4] Found pairing with 226B3592-2307-44C6-9C4D-E087492C6239
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> HomeKit: [Client 6] Pair Verify Step 2/2
>>> HomeKit: [Client 6] Found pairing with 226B3592-2307-44C6-9C4D-E087492C6239
>>> HomeKit: [Client 6] Verification successful, secure session established
>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> HomeKit: [Client 5] Found pairing with 226B3592-2307-44C6-9C4D-E087492C6239
>>> HomeKit: [Client 5] Verification successful, secure session established
>>> HomeKit: Got new client connection: 7 from 192.168.1.125
>>> HomeKit: [Client 6] Get Accessories
HAA > Getter
HAA > Getter
HAA > Getter
>>> HomeKit: [Client 4] Get Accessories
HAA > Getter
HAA > Getter
HAA > Getter
>>> HomeKit: [Client 7] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Get Accessories
HAA > Getter
HAA > Getter
HAA > Getter
>>> HomeKit: [Client 6] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 7] Pair Verify Step 2/2
>>> HomeKit: [Client 7] Found pairing with BDD9D4D4-7949-4C18-9505-1222EF2FA05C
>>> HomeKit: [Client 7] Verification successful, secure session established
>>> HomeKit: [Client 7] Get Accessories
HAA > Getter
HAA > Getter
HAA > Getter
>>> HomeKit: [Client 7] Update Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Get Characteristics
HAA > Getter
HAA > Getter
HAA > Getter
HAA > GD sensor: 0
>>> HomeKit: [Client 5] Update Characteristics
HAA > Setter GD
HAA > Digital output GPIO 12 -> 1
HAA > Autoswitch digital output GPIO 12 -> 0
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Update Characteristics

when I shorted GPIO14 with GND nothing appear in logs, when I press physical button nothing appear in logs, when I press button in apple home app appear this line:
>>> HomeKit: [Client 5] Update Characteristics

pilot1981, 2019-12-25:
I also attach log file. I hope this could be useful...

[screenlog.0.zip](https://github.com/RavenSystem/esp-homekit-devices/files/4000759/screenlog.0.zip)

RavenSystem, 2019-12-25:
> when I shorted GPIO14 with GND nothing appear in logs

Check your hardware. JSON is right with GPIO-14 setup.

> when I press physical button nothing appear in logs

This is right, because built-in button is only used to enter setup mode in that JSON.

pilot1981, 2019-12-25:
I have same problem on 2 different sonoff...same GPIO14 worked fine on RC

Ok, how I can add physical button to open garage and use dummy switch to enter in setup mode? When I use dummy switch I have to press only one time or 8 times?

RavenSystem, 2019-12-25:
> I have same problem on 2 different sonoff...same GPIO14 worked fine on RC

Can you put a photo of your device where GPIO 14 is showed, please?

> Ok, how I can add physical button to open garage and use dummy switch to enter in setup mode?

I told you before:
This is JSON with inverted sensor, built-in button to operate door, and HomeKit switch to enter setup mode:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":14}],"f4":[{"g":14,"t":0}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"b":[{"g":0}]},{"1":{"s":[{"a":1}]}}]}

>  When I use dummy switch I have to press only one time or 8 times?

If dummy switch has no actions, 8 times. If switch has a system action to enter setup mode (like JSON above), once.

pilot1981, 2019-12-25:
[image]

RavenSystem, 2019-12-25:
¬¬ That is not GPIO 14. It is GPIO 3 (RX pin)!!!

Please, next time ensure that you have the right info.

Here is your JSON, and reset homekit ID again:
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":3,"t":0}],"f4":[{"g":3}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"b":[{"g":0}]},{"1":{"s":[{"a":1}]}}]}`

closed 2019-12-25 by RavenSystem

pilot1981, 2019-12-26:
In this JSON:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2}}]}

How I can enter in setup mode? I tried physical button 8 times but nothing...

pilot1981, 2019-12-26:
Hi,
thanks @RavenSystem, now close sensor work fine but I have another problem, I go to explain context of problem.

1) garage door is opened
2) I send closing command, relays start well and garage begin closing
3) if security sensor send signal to garage door built-in mechanism, garage door stop and remain partially opened, in home app I have garage door in closing status
4) the problem is I'm not able to start again relays using home app or physical button, so it's not possibile to close or open garage door.

in RC I'm able to start again relays independently from garage door status, can we have same feature also in HAA?

pilot1981, 2019-12-28:
Hi @RavenSystem, any news for me?
Can you re-open this issue?

pilot1981, 2019-12-29:
Hi @RavenSystem any news?

pilot1981, 2019-12-31:
@RavenSystem howto enable stop-step feature? without this feature I think garage door device type is un useful when you have some problem  on open/close sensor...you agree?

Andre000Dias, 2020-05-06:
@RavenSystem 
Spanish / English pem message
________________________________________
ENGLISH:
I liked your profile, I have a lot to learn from you;)
I followed this discussion from the beginning ..

I copied the following code:

{"c": {"l": 13, "b": [{"g": 0, "t": 5}]}, "a": [{"t": 40, "d": 18 , "f3": [{"g": 3, "t": 0}], "f4": [{"g": 3}], "0": {"r": [{"g": 12, "v": 1, "i": 0.5}]}, "1": {"a": 0}, "2": {"r": [{"g": 12, "i": 1.5}, {"g": 12, "v": 1, "i": 0.5}, {"g": 12, "v": 1, "i": 2}]}, "3": { "a": 2}, "b": [{"g": 0}]}, {"1": {"s": [{"a": 1}]}}]}

in this case, the gate just opens or closes, does not make any stops ...
 would it be possible to help me with the code with stops? that is ,

1. (initial state CLOSED), giving a pulse the gate starts to open.
2. giving a pulse while he is opening the gate stops.
3. another pulse, it has just opened up to full opening.
4. the same for closing ..
______
if you did not give the pulse in the middle of the opening / closing the gate would normally run without stops.

_____________________________________________________________________
ESPANOL:
Me gustó tu perfil, tengo mucho que aprender de ti;)
Seguí esta discusión desde el principio.

Copié el siguiente código:

{"c": {"l": 13, "b": [{"g": 0, "t": 5}]}, "a": [{"t": 40, "d": 18 , "f3": [{"g": 3, "t": 0}], "f4": [{"g": 3}], "0": {"r": [{"g": 12, "v": 1, "i": 0.5}]}, "1": {"a": 0}, "2": {"r": [{"g": 12, "i": 1.5}, {"g": 12, "v": 1, "i": 0.5}, {"g": 12, "v": 1, "i": 2}]}, "3": { "a": 2}, "b": [{"g": 0}]}, {"1": {"s": [{"a": 1}]}}]}

en este caso, la puerta simplemente se abre o se cierra, no se detiene ...
 ¿Sería posible ayudarme con el código con paradas? o sea ,

1. (estado inicial CERRADO), dando un pulso a la puerta comienza a abrirse.
2. dando un pulso mientras abre la puerta se detiene.
3. otro pulso, se acaba de abrir a la apertura total.
4. lo mismo para cerrar ..
______
si no le dio el pulso en medio de la apertura / cierre, la puerta normalmente funcionaría sin paradas.

Danilo0609, 2023-06-11:
tengo el mismo problema. la apertura funciona perfecto. pero al cerrar se cierra en forma pacial. se detiene. tengo que insistir para que se cierre. utilice el json {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":14,"t":0}],"f4":[{"g":14}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2}}]}. 

necesitaria que el porton se abra con un toque, se pause con un toque, se cierre con un toque y se pause tambien con un toque. me podran ayudar?