RavenSystem/esp-homekit-devices #686: Upgrading from RavenCore to HAA

GPL71, 2019-12-23:
**IMPORTANT**
This guide allows the upgrade from RavenCore firmware to the new HAA.
Whatever RavenCore firmware is on the device, the update will install version 1.3.0 and then the upgrade to HAA.
Upgrade to HAA is strongly recommended because it solves the ESP8266 HomeKit library bug.
Furthermore, now all the functions available in RavenCore have been imported into HAA and more customizable.

To upgrade follow the steps:

1. Create or copy from RavenSystem examples page the suitable JSON configuration for wanted accessory. 
2. Open EVE and enter SETUP mode
3. Take note of device IP address you'll update
4. Set FIRMWARE UPDATE to "YES", set REBOOT to "YES" and close EVE.

[image]

5. Wait approximately 5-7 minutes. That is necessary time to download and installation HAA firmware
6. Open browser and enter the IP address of the device
7. When configuration web page opens, enter following information:

- JSON configuration
- Reset Homekit ID
- Save

8. Web page closes and configuration begins
9. Open HOME app and remove the old device
10. Add new device
11. Type the code or read QR Code
12. Wait for 30 to 60 seconds and device will added to HomeKit

NOTE: If upgraded device include only one accessory (for example Shelly 1 as 1 object control), perform Reset Homekit ID in point 7 or points 9 to 12 are not necessary.

Dirk23, 2019-12-25:
Thank you very much for this manual. Unfortunately I cannot upgrade my Sonoff Dual R2 to HAA firmware.  I have followed every step up to point 5 and I am not getting anywhere. There is no website under the IP address of the device.  I also have a point called Firmware HAA in my settings (in the Eve App) which is mentioned in the manual. Does this switch possibly have to be set as well? 
[image]

GPL71, 2019-12-29:
@Dirk23 toggle on REBOOT too and continue closing HOME app.

Dirk23, 2019-12-29:
> @Dirk23 toggle on REBOOT too and continue closing HOME app.

I found the Problem, it seems that my firewall blocked Something the HAA needs to load.

GPL71, 2019-12-31:
> > @Dirk23 toggle on REBOOT too and continue closing HOME app.
> 
> I found the Problem, it seems that my firewall blocked Something the HAA needs to load.

can you disable it?

Dirk23, 2019-12-31:
> > > @Dirk23 toggle on REBOOT too and continue closing HOME app.
> > 
> > 
> > I found the Problem, it seems that my firewall blocked Something the HAA needs to load.
> 
> can you disable it?

Sure, i disabled the Firewall and everything works fine now.

closed 2020-02-19 by stale[bot]

Quoshak, 2020-03-02:
> 
>     2. Open EVE and enter SETUP mode
> 

How do you enter setup mode in EVE app? I cannot find anything that would bring me features that I can see on your screenshot.

frazer81, 2020-03-08:
> 
> 
> > ```
> > 2. Open EVE and enter SETUP mode
> > ```
> 
> How do you enter setup mode in EVE app? I cannot find anything that would bring me features that I can see on your screenshot.

Hi - i'm also looking for the same interface on the EVE app - is anyone able to provide guidance on how to access the setup page within? It something we need to enable within the JSON to access perhaps? Appreciate the assistance!!

asbesto01, 2020-03-22:
> **IMPORTANT**
> This guide allows upgrade from RavenCore firmware to the new HAA.
> Whatever RavenCore firmware is on the device, the update will install version 1.3.0 and then the upgrade to HAA.
> Upgrade to HAA is strongly recommended because it solves the ESP8266 HomeKit library bug.
> Furthermore, now all the functions available in RavenCore have been imported into HAA and more customizable.
> 
> To upgrade follow the steps:
> 
> 1. Create or copy from RavenSystem examples page the suitable JSON configuration for wanted accessory.
> 2. Open EVE and enter SETUP mode
> 3. Take note of device IP address you'll update
> 4. Set FIRMWARE UPDATE to "YES", set REBOOT to "YES" and close EVE.
> 
> [image]
> 
> 1. Wait approximately 5-7 minutes. That time is necessary for HAA download and installation
> 2. Open browser and enter the device IP address
> 3. When configuration web page opens, enter following information:
> 
> * JSON configuration
> * Reset Homekit ID
> * Save
> 
> 1. Web page closes and configuration begins
> 2. Open HOME app and remove the old device
> 3. Add new device
> 4. Type the code or read QR Code
> 5. Wait for 30 to 60 seconds and device will added to HomeKit
> 
> NOTE: If upgraded device include only one accessory (for example Shelly 1 as 1 accessory), perform Reset Homekit ID in point 7 or points 9 to 12 are not necessary.

after point 4, device reboots but does not upgrade the ravencore to 1.3, nor download the haa.
I am now on ravencore 1.1.1

any help?

pilot1981, 2020-03-22:
There is a problem with GITHUB ORA procedure...