RavenSystem/esp-homekit-devices #687: problem to insert dummy switch in JSON template of shelly 1 for light HAA

pilot1981, 2019-12-24:
Hi,
I tried to use this JSON template for light with shelly 1:

{"a":[{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]}]}

and I added dummy switch for setup mode; my new JSON is this:

{"a":[{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"1":{"s":[{"a":1}]}}]}

but with this new JSON, I'm not able to turn on the light using both button...

After trying buttons sometimes, device begin unresponsive in home app...

To enter again in setup mode, I quickly toggled physical button 8 times, but device don't start web page...

Adding string {"1":{"s":[{"a":1}]}} isn't possibile use physical button to enter in setup mode?

Tomasz-Bochenek, 2019-12-25:
I had similar issue and my problem was the wrong configuration of the initial configuration of the switch to enter setup mode (missing "p": 0).  What works for me is:
{
  "c": { "b": [{ "g": 5, "p": 0, "t": 5 }] },
  "a": [{
    "t": 1,
    "0": { "r": [{ "g": 4, "v": 0 }] },
    "1": { "r": [{ "g": 4, "v": 1 }] },
    "b": [ { "g": 5, "p": 0, "t": 1 }, { "g": 5,  "p": 0, "t": 0 }]
  }]
}

pilot1981, 2019-12-25:
I changed JSON but I’m not able to add accessory in home app; I removed from home apps and I done reset HomeKit ID, if I use QR CODE home app say always accessory already added...if I use normal code I don’t find any accessory in home app

What I can do?

closed 2020-01-10 by RavenSystem