RavenSystem/esp-homekit-devices #688: Template Help: Curtain Motor

tonysprenk, 2019-12-24:
This is the device: https://s.click.aliexpress.com/e/n3g3Q1For
More info [here](https://github.com/arendst/Tasmota/wiki/Zemismart-Curtain-Motor).

Any help would be greatly appreciated!

tonysprenk, 2019-12-24:
https://tasmota.github.io/docs/#/TuyaMCU?id=curtain-motors this might help too!

RavenSystem, 2019-12-25:
Sorry, it is not compatible with HAA because esp chip does not manage curtain directly. It uses serial commands to communicate with the chip that manages it.

tonysprenk, 2019-12-25:
Thats really bad news... Any plans on adding this feature into HAA? Tasmota seems to have it implemented, but I love HAA as I don't need a bridge, and it's very stable for my other accessories... I think more info on how Tasmota achieves this can be found [here](https://tasmota.github.io/docs/#/TuyaMCU).

RavenSystem, 2019-12-25:
Maybe in the future, but not soon.

bj-tracer, 2019-12-26:
You may use other curtain drives with the direct motor control and any dual relay with esp8266 controller.
If you want to use Zemismart motor than you need a tiny controller that may convert hi/low signals from esp8266 with HAA to serial comands.

bj-tracer, 2019-12-26:
or... you can remove Zemismart logic and inser any dual relay module with esp8266 ;)

tonysprenk, 2019-12-26:
Yes I’m researching this. It has a rj45 connection for dry connecting a low voltage relay. I guess something like this (https://s.click.aliexpress.com/e/FzW1z1iXv) should work?

tonysprenk, 2019-12-26:
[image]

tonysprenk, 2019-12-26:
I’d like to keep the internal components intact as I use the RF remote as wel as the “manual open and close” functionality.

RavenSystem, 2019-12-26:
Using externals controls (Like RF) is not good, because HomeKit loses position sync.

Other way could be to build a device that works over RF remote, connecting relays to RF buttons, to open, close and stop. You lose RF remote, but as I say, using controls outside HomeKit is not a good idea.

tonysprenk, 2019-12-26:
Hmmm okay, maybe I should just resort to using the homeassistant Tuya component as it has the ability to get the state, unfortunately I’ll be connected to a Chinese server. Thanks for all your suggestions!

bj-tracer, 2019-12-27:
You can use RF that connected to esp8266 board and working like an external switches.
Then Zemismart board will to drive the motor only.

So you need to make the stick that will be connect to the Zemismart drive trough RJ45 with 1/2 logic (for example).
Unfortunately there is no power in this RJ45 connection. But you can use for it USB-port ;)

closed 2020-01-10 by RavenSystem