RavenSystem/esp-homekit-devices #689: Collection of TESTED json

GPL71, 2019-12-24:
A collection of tested json.
Please add your.
FORMAT:
- Device
- Accessory description
- Json

GPL71, 2019-12-24:
- 2 Channels Relay
- 1 Lock Mechanism, 1 inching relay (0.6 sec both)
- Json:
{"c":{"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1}]},"1":{"r":[{"g":12,"v":0}]},"t":4,"i":0.6,"s":1,"b":[{"g":0,"t":1}]},{"0":{"r":[{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1}]},"i":0.6,"s":0,"b":[{"g":0,"t":2}]}]}

GPL71, 2019-12-24:
- SONOFF Mini
- Temperature sensor (GPIO 4, DS18B20), Motion Detection (GPIO 2, Mini PIR), Lock Mechanism (GPIO 12 Relay, GPIO0 Button, default Locked)
- Json:
   {"c":{"f":100,"b":[{"g":0,"t":5}]},"a":[{"t":22,"g":4,"n":3,"b":[]},{"t":12,"f0":[{"g":2,"t":0}],"f1": 
   [{"g":2}],"s":0},{"0":{"r":[{"g":12,"i":0.3,"v":1}]},"1":{"r":[{"g":12,"v":0}]},"t":4,"i":3,"b":[{"g":0}],"s":1}, 
   {"1":{"s":[{"a":1}]},"b":[],"s":0}]}

GPL71, 2019-12-24:
- 2 Channels Relay
- 2 inching Relay (0.6 sec both)
- Json:
  {"c":{"b":[{"g":0,"t":5}]},"a":[{"i":0.6,"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"b": 
  [{"g":0,"t":1}]},{"i":0.6,"0":{"r":[{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":0,"t":2}]}]}

This Json is by RavenSystem

GPL71, 2019-12-24:
- SONOFF 4ch Pro R2
- Setup Mode (GPIO0), 2 Switches, 1 Outlet, 1 Thermostat (GPIO 2,DHT22)
- Json
{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14}]},
{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":2,"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":9}]},{"t":21,"s":5,"g":2,"0":{"r":[{"g":4}]},"1":{"a":0},"3":{"r":[{"g":4,"v":1}]},"5":{"a":0}}

GPL71, 2019-12-24:
- SONOFF 4ch Pro R2
- Setup Mode (GPIO0), 2 Inching relay, 2 Switches
{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"t":4,"i":0.4,"0":{"r":[{"g":12,"v":1}]},"1":{"r":[{"g":12,"v":0}]},"b":[{"g":0}]},{"t":4,"i":0.4,"0":{"r":[{"g":5,"v":1}]},"1":{"r":[{"g":5,"v":0}]},"b":[{"g":9}]},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":10}]},{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14}]}]}

GPL71, 2019-12-24:
- SONOFF 4ch Pro R2
- Setup Mode (GPIO0), 4 Water Valve (Sprinkler)
- Json:
  {"c":{"o":0,"m":60,"l":13,"i":0,"f":10,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1}]},"1":{"r": 
  [{"g":12,"v":0}]},"t":20,"b":[{"g":0,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":5,"v":1}]},"1": 
  {"r":[{"g":5,"v":0}]},"t":20,"b":[{"g":9,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":4,"v":1}]},"1": 
  {"r":[{"g":4,"v":0}]},"t":20,"b":[{"g":10,"t":1,"p":1}],"s":0,"w":0,"d":0,"f0":[]},{"0":{"r": 
  [{"g":15,"v":1}]},"1":{"r":[{"g":16,"v":0}]},"t":20,"b":[{"g":14,"t":1,"p":1}],"s":0,"w":0,"d":0,"f0":[]}]}

GPL71, 2019-12-24:
- 2 Channels
- Setup Mode (GPIO0), 1 Lock Mechanism (GPIO12), 1 Switch (GPIO5)
- Json:
   {"c":{"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1}]},"1":{"r":[{"g":12,"v":0}]},"t":4,"i":0.5,"s":1,"b": 
   [{"g":0,"t":1}]},{"0":{"r":[{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1}]},"s":0,"b":[{"g":0,"t":2}]}]}

GPL71, 2019-12-24:
- SONOFF SV
- Setup Mode (GPIO0), Garage Door, 1 magnet contact (GPIO14, fully open) and 1 magnet contact 
   (GPIO4, fully closed). Fully opened in 31 seconds
- Json:
  {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5}, 
  {"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"t":40,"d":31,"f3":[{"g":14,"t":0}],"f4": 
  [{"g":14}],"f2":[{"g":4,"t":0}],"f5":[{"g":4}]}]}

peros550, 2019-12-27:
- Blitzwolf BW-LT11 Led strip

- GPIO0: Button1,GPIO4: PWM1, GPIO5: PWM4, GPIO12: PWM2, GPIO14: PWM3

- JSON: `{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":4,"g":12,"v":14,"w":5,"b":[{"g":0}]}]}`

skv6, 2019-12-28:
- Blitzwolf BW-SHP2 outlet 

-  GPIO1: led, GPIO3: button, GPIO14: relay 

-  JSON:
 {"c":{"l":1,"b":[{"g":3,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1}]},"b":[{"g":3}]}]}

pawelosin, 2019-12-30:
- **Power Strip SA-P402A** with 5 function button

-  Setup Mode (GPIO0), 5 Outlets (including USB)

- JSON
{"c":{"l":3,"i":1,"b":[{"g":0,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":1}]},{"t":2,"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1}]},"b":[{"g":1,"t":2}]},{"t":2,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":1,"t":3}]},{"t":2,"0":{"r":[{"g":13}]},"1":{"r":[{"g":13,"v":1}]},"b":[{"g":1,"t":4}]},{"t":2,"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":1,"t":5}]}]}

pawelosin, 2019-12-30:
- **Sonoff TH 16**

- Setup Mode (GPIO0), Thermostat (GPIO14, DS18B20, max temperature 75 degrees Celsius)

- JSON 
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":21,"n":3,"s":5,"x":75,"g":14,"0":{"r":[{"g":12}]},"1":{"a":0},"3":{"r":[{"g":12,"v":1}]},"b":[{"g":0}],"5":{"a":0}}]}

pawelosin, 2019-12-30:
**Switch, temperature sensor and leak sensor**

- Setup Mode (GPIO0), termperature sensor (GPIO5, DS18B20), leak sensor (GPIO14)

- JSON
{"c":{"l":2,"i":1,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":22,"n":3,"g":5},{"t":7,"f1":[{"g":14,"t":1}],"f0":[{"g":14,"t":0}]}]}

GPL71, 2019-12-31:
- Shelly 1
- Outlet controlled by external switch (outlet icon in HOME app)
- JSON:
   {"a":[{"t":2,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]}]}

dido92, 2020-01-05:
Please I need early json string for my Shelly 2.5 windows covering whit external push button. 😩 i have a big problem 😩 please help me

diodo157, 2020-01-07:
- Multi Presa houzetek
This Json works fine, the button above is set as stateless

```
{"c":{"o":1,"l": 3,"i":1,"b":[{"g":1,"t":5}]},
"a":[
{"t":2,"0":{"r":[{"g": 12,"v":0}]},"1":{"r":[{"g":12,"v":1}]}},
{"t":2,"0":{"r":[{"g":14,"v":0}]},"1":{"r":[{"g": 14,"v":1}]}},
{"t":2,"0":{"r":[{"g":13,"v":0}]},"1":{"r":[{"g":13,"v":1}]}},
{"t":2,"0":{"r":[{"g":4,"v":0}]},"1":{"r":[{"g":4,"v":1}]}},
{"t":2,"0":{"r":[{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1}]}},

{"t":3,"f0":[{"g":1}],"f1":[{"g":1,"t":2}],"f2":[{"g":1,"t":3}]}
]}
```
[image]

mikebatnz, 2020-01-08:
- Mirabella Genio 1002339 (BC) or 1002338 (ES)

- https://mirabellagenio.net.au/es-%2F-bc-warm-rgb-specs

`{"a":[{"t":30,"r":4,"g":12,"v":14,"w":5,"s":1}]}`

mikebatnz, 2020-01-08:
- Brilliant Lighting Smart WiFi Plug
NOTE: Energy monitoring not supported (yet)

- https://www.officeworks.com.au/shop/officeworks/p/brilliant-lighting-smart-wifi-plug-with-energy-monitoring-bl20925

`{"c":{"l":13,"b":[{"g":13,"t":5}],"i":1},"a":[{"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1}]},"t":2,"b":[{"g":3,"t":1}],"s":5,"f0":[],"f1":[]}]}`

mikebatnz, 2020-01-08:
- Kogan SmarterHome 5W Smart Dimmable LED Filament Bulb ST-20 (E27)

- https://www.kogan.com/au/buy/kogan-smarterhome-5w-smart-dimmable-led-filament-bulb-st-20-e27/

`{"a":[{"t":30,"w":14,"s":1}]}`

mikebatnz, 2020-01-08:
- Mirabella Genio WiFI LED G95 #1002747 (ES)

`{"a":[{"t":30,"v":14,"w":12,"s":1}]}`

GPL71, 2020-01-12:
- 4 channel switch and relay and dummy switch
- Sonoff 4CH Pro R2 or wemos or NodeMCU
- JSON:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14}],"s":0},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}],"s":0},{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":9}],"s":0},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":10}],"s":0},{"t":1,"b":[],"s":0,"i":0.6}]}

peterduyts, 2020-01-15:
Shelly 1 pulse switch
gives a 1 sec. pulse, then returns to the off position

{"a":[
{
"t": 1,
"d": 1,
"0":{"r":[{"g":4}]},
"1":{"r":[{"g":4,"v":1,"i":1}]},
"b":[{"g":5,"p":0},{"g":5,"t":0}]}]}

peros550, 2020-01-19:
Blitzwolf BW-LT21
tested: `{"c":{"n":"Led-BW-LT21"},"a":[{"t":30,"r":14,"g":12,"v":5,"w":15}]}`

GPL71, 2020-01-20:
- 1 switch and 1 button, 2 relays 
- Shelly 2.5
- JSON:
{"c":{"l":0,"b":[{"g":2,"t":5}]},"a":[{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0},                            {"g":5,"t":0}]},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":13,"p":0,"i":1}]}]}

md-zaikin, 2020-01-26:
ESP-01S + Relay v4 
Relay GPIO - 0 - defolt, Button -GPOI 2 ( on defoult butten on relay conectet to RST!)
{"c":{"l":13,"b":[{"g":2,"t":5}],"i":0,"n":"ESP_01S"},"a":[{"0":{"r":[{"g":0,"v":1}]},"1":{"r":[{"g":0,"v":0}]},"b":[{"g":2}],"s":0}]}

vglucas, 2020-01-29:
ESP01 with Relay v.4 GPIO0
add switch GPIO3

json
{"c":{"l":13},"a":[{"t":1,"0":{"r":[{"g":0,"v":0}]},"1":{"r":[{"g":0,"v":1}]},"b":[{"g":3,"t":1},{"g":3,"t":0}]}]}

[image]

[image]

belkone, 2020-02-03:
**LSC Smart Connect Smart Power Plug** from [Action](http://action.com)

`{"c":{"l":4,"b":[{"g":14,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":14}]}]}`

rasod, 2020-02-04:
- **Smart Switch Under Various Brands**
- https://www.amazon.com/dp/B07B622LYF/
- https://www.amazon.ca/dp/B07T9DP4PJ/
- {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}

[image]

pilot1981, 2020-02-11:
- Sonoff Basic R2 cutted track
- GPIO3 for closed sensor
- {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":3,"t":0}],"f4":[{"g":3}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"b":[{"g":0}]},{"1":{"s":[{"a":1}]}}]}

platinman, 2020-02-19:
Blitzwolf bw-ss1 

- GPIO4: led,
- GPIO13: button, 
- GPIO5: relay 
Switch
{"c":{"l":4,"b":[{"g":13,"t":5}]},"a":[{"t":1,"s":5,"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":13}]}]}

rasod, 2020-02-20:
### Teckin SL02 LED Strip
[image]

- https://www.amazon.co.uk/TECKIN-Control-Changing-Flexible-Decoration/dp/B07YS6V1BD
- https://www.amazon.com/Compatible-TECKIN-Million-Changing-Decoration/dp/B07YS5T23F
- https://www.amazon.ca/TECKIN-Million-Changing-Control-Decoration/dp/B07YSB3242
- {"a":[{"t":30,"r":4,"g":12,"v":14,"s":1}]}
- IR remote not configured

mriksman, 2020-02-22:
https://www.aliexpress.com/item/4000615022697.html
AVATTO US/AU Wall Switch
SYS-WS 01/02/03
[image]

```
GPIO13			Relay1
GPIO15			Relay2
GPIO4			Relay3
GPIO16			Button1
GPIO5			Button2
GPIO3 (RXD0)    	Button3
GPIO12			LED1
GPIO1 (TXD0)    	LED2
GPIO14 			LED3
GPIO0			Status LED
```
For a 2-gang switch;
```
{
	"c": {
		"l": 0,
		"b": [{"g": 16,	"t": 5}]
	},
	"a": [
		{
			"t": 1,
			"0": {"r": [{"g": 13, "v": 0}, {"g": 12, "v": 1}]},
			"1": {"r": [{"g": 13, "v": 1}, {"g": 12, "v": 0}]},
			"b": [{"g": 16,	"t": 1}]
		},
		{
			"t": 1,
			"0": {"r": [{"g": 4, "v": 0}, {"g": 14, "v": 1}]},
			"1": {"r": [{"g": 4, "v": 1}, {"g": 14, "v": 0}]},
			"b": [{"g": 1,	"t": 1}]
		}
	]
}

```

princefisher, 2020-02-27:
## Device(s)

All Smart Plugs with Tuya ESP Chip tested:

- Worked with [Gosund](https://www.amazon.com/dp/B079MFTYMV/ref=psdc_6291368011_t1_B079KYVRN8) --> WP3 Model
- Worked with [Gosund "Upgraded Version"](https://www.amazon.com/gp/product/B07GRLQV47/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) --> WP5 Model
- Worked with [TanTan](https://www.amazon.com/gp/product/B079KYVRN8/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1) --> WP3 Model

## Accessory description

Wifi Smart Plug GPIO Configuration
- GPIO4: Button
- GPIO12: LED 2i (red)
- GPIO13: LED 1i (blue)
- GPIO14: Relay

## JSON

- This configuration will use the blue LED for configuration changes and status changes.
- The red LED is used to signal state, 
  - red LED on is power is on, 
  - red LED off is power is off.
- The button will turn the plug on an off. 
- Pressing the button 8 times will reboot in configuration mode.
- the name is specified in `"n"` -> `"name-your-outlet"`
- the initial state is `off`

Formatted:
```
{
  "c": {
    "l": 13,
    "n": "name-your-outlet",
    "b": [
      {
        "g": 4,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12,
            "v": 1
          },
          {
            "g": 14,
            "v": 0
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 0
          },
          {
            "g": 14,
            "v": 1
          }
        ]
      },
      "b": [
        {
          "g": 4,
          "t": 1,
          "p": 1
        }
      ],
      "s": 0
    }
  ]
}
```

One-Liner:
```
{"c":{"l":13,"n":"name-your-outlet","b":[{"g":4,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1},{"g":14,"v":0}]},"1":{"r":[{"g":12,"v":0},{"g":14,"v":1}]},"b":[{"g":4,"t":1,"p":1}],"s":0}]}
```

## Notes
- flashed with [tuya-convert](https://github.com/ct-Open-Source/tuya-convert) within seconds using a stock Raspberry-Pi Zero W

pilot1981, 2020-03-23:
Shelly 1 used to control water valve; on switch of GPIO5 I attached external physical button stateless to control manually water valve; I added also a dummy switch to enter in setup mode:

{"a":[{"t":20,"d":3600,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0}]},{"i":0.5,"1":{"s":[{"a":1}]}}]}

Crazor, 2020-04-04:
Teckin SP22
Smart Plug

`{"c":{"l":3,"b":[{"g":1,"t":5}]},"a":[{"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1}]},"t":2,"b":[{"g":1}]}]}`

pilot1981, 2020-04-06:
Shelly 2.5
2 switch with initial state
Dummy switch to enter in setup mode

{"c":{"l":0,"i":1},"a":[{"s":5,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":13,"p":0},{"g":13,"t":0}]},{"s":5,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"1":{"s":[{"a":1}]}}]}

TriderG75, 2020-04-15:
Universal IR device
https://gearbest.app.link/7kIjGiVCz5 (Tested Type B and Type C versions)
https://a.aliexpress.com/_d7NKf18 (Not tested)
https://s.click.aliexpress.com/e/_d7qlxvI (Tested)
(Thanks to carminedelia, micheletorrini and PSB on Discord)

This json is for managing a Samsung TV with an IR device (see links above).
IR commands can be found here : https://github.com/RavenSystem/esp-homekit-devices/wiki/IR-Codes-Database.
Also available Philips and LG IR commands.
Json has also an IP function to know if smart Tv is On or Off using ping and a dummy switch that allows you to enter in setup mode by pressing it one time.

Json:
`{"c":{"t":14,"p":"J*J*AaAaAaDFAa"},"a":[{"t":60,"xa":0,"i":[{"n":"MENU","0":{"i": [{"r":2,"c":"CeCfAaBcAaAbC"}]}},{"n":"SOURCE","0":{"i":[{"r":2,"c":"CeCeAhG"}]}},{"n":"MUTE","0":{"i":[{"r":2,"c":"CeCeDhD"}]}}],"0":{"i":[{"r":4,"c":"CeCfAfAaF"}]},"1":{"i":[{"r":4,"c":"CeCfAfAaF"}]},"6":{"i":[{"r":2,"c":"CeCjBaEbA"}]},"7":{"i":[{"r":2,"c":"CeCeAdBbDbA"}]},"8":{"i":[{"r":2,"c":"CeCeAaAbBbAaBbA"}]},"9":{"i":[{"r":2,"c":"CeCfAcBaAaCbA"}]},"10":{"i":[{"r":2,"c":"CeChAaBaCaAbA"}]},"11":{"i":[{"r":2,"c":"CeChBaAaCbAaA"}]},"13":{"i":[{"r":2,"c":"CeCeBaAaBcAaAbA"}]},"17":{"i":[{"r":2,"c":"CeCeEhC"}]},"22":{"i":[{"r":2,"c":"CeCeChE"}]},"23":{"i":[{"r":2,"c":"CeCeBaAfAaD"}]},"q0":[{"h":"192.168.1.100","r":0}],"q1":[{"h":"192.168.1.100","r":1}]},{"i":0.5,"1":{"s":[{"a":1}]}}]}`

Tv_actions
[image]

Gocomma Type B
[image]
[image]
[image]

Cogomma Type C
[image]
[image]
[image]

Avatto S07
[image]
[image]

TriderG75, 2020-04-15:
- SPETU Wifi Water Smart Valve 
https://a.aliexpress.com/_dYF5qxo (Tested. Thanks to micheletorrini on Discord)

- Json for managing a smart water valve. Used the 2 onboard leds (red and blue). It has a dummy switch too that allows you to enter in setup mode by pressing it one time. Parameter `"w":`
can be changed with 0,1,2,3 values for different valve types. Added `"d":0` to disable in app HomeKit valve timer.

- Json:
`{"c":{"l":5},"a":[{"t":20,"w":3,"d":0,"s":5,"0":{"r":[{"g":12},{"g":4,"v":0}]},"1":{"r":[{"g":12,"v":1},{"g":4,"v":1}]},"b":[{"g":13,"p":1,"t":1}]},{"i":0.5,"1":{"s":[{"a":1}]}}]}`

Gpio list:
https://templates.blakadder.com/jinvoo_SM-AW713-v2.html

[image]

TriderG75, 2020-04-15:
- Jinvoo Smart Wi-Fi Touch panel US 3
https://www.amazon.it/dp/B07NJM13RB/ref=cm_sw_r_cp_api_i_QxTLEbDJ5SZC7 (Tested. Thanks to branda92 on Discord)

- The json manages the 3 gangs of the device and adds a dummy switch too that allows you to enter in setup mode by pressing it once.
WARNING : Problems with the flashing procedure (see photos below)

- Json:
`{"c":{"l":0,"i":1,"b":[{"g":12,"t":5}]},"a":[{"0":{"r":[{"g":13}]},"1":{"r":[{"g":13,"v":1}]},"b":[{"g":12}]},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":3}]},{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5}]},{"i":0.5,"1":{"s":[{"a":1}]}}]}`

Gpio list:
https://templates.blakadder.com/jinvoo_SM-SW101-3.html

[image]

[image]

[image]

[image]

[image]

Note: The flashing problem was caused by the chip U4 presence. This chip manages the 3 capacitive touch buttons and has its serial pin connected to the TYWE3S Rx pin. So we had to desolder U4, flash TYWE3S and then solder U4 again. I suppose that disconnecting pins 5 and 6 (maybe only pin 5) is enough.

TriderG75, 2020-04-15:
- Electrodragon Wifi IoT SPDT Relay Board Based on ESP8266
https://www.electrodragon.com/product/wifi-iot-relay-board-spdt-based-esp8266/
(Tested. Thanks to jvog85 on Discord)

- The json manages a windows covering and also adds a dummy switch that allows you to enter in setup mode by pressing it once. Gpio4/5 were connected to 3 state switch (up/neutral/down)

- Json:
`{"c":{"l":16,"b":[{"g":2,"t":5}]},"a":[{"t":45,"o":20,"c":19,"f":70,"0":{"r":[{"g":13},{"g":12,"v":1}]},"1":{"r":[{"g":13,"v":1},{"g":12}]},"2":{"r":[{"g":13},{"g":12}]},"3":{"r":[{"g":13},{"g":12,"i":1}]},"4":{"r":[{"g":13,"i":1},{"g":12}]},"f0":[{"g":5,"t":0}],"f1":[{"g":4,"t":0}],"f2":[{"g":5,"t":1},{"g":4,"t":1}]},{"i":0.5,"1":{"s":[{"a":1}]}}]}`

Gpio list:
https://templates.blakadder.com/electrodragon_board.html

[image]

PsbDxb, 2020-04-19:
- Tuya Smart Switch, DS102 (3 gang)
https://s.click.aliexpress.com/e/_dTvq6e2

- The json manages the 3 gangs of the device, and goes into setup mode by holding the left side switch for 5 seconds

- Json
{"c": {"l":0,"b":[{"g":12,"t":5}]},"a":[ {"0":{"r":[{"g":15}, {"g":1,"v":1}]},"1":{"r":[{"g":15,"v":1},{"g":1,"v":0}]}, "b":[{"g":5}]},{"0":{"r":[{"g":13}, {"g":16,"v":1}]},"1":{"r":[{"g":13,"v":1},{"g":16,"v":0}]}, "b":[{"g":12}]},{"0":{"r":[{"g":4}, {"g":14,"v":1}]},"1":{"r":[{"g":4,"v":1}, {"g":14,"v":0}]}, "b":[{"g":3}]}]}

GPIO list:
https://templates.blakadder.com/DS-102_3.html

[image]
[image]

pilot1981, 2020-04-20:
Shelly 1
1 switch for lightbulb with initial state and dummy switch to enter in setup mode:

{"a":[{"s":5,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"i":0.5,"1":{"s":[{"a":1}]}}]}

sercitecotf, 2020-04-23:
SONOFF 4ch 
4 Water Valve with timer, made with haajc v0.6
Json:
{"c":{"o":0,"l":13,"i":0,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":20,"b":[{"g":0,"t":1,"p":0}],"s":0,"w":0,"d":3600,"f0":[]},{"0":{"r":[{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1}]},"t":20,"b":[{"g":9,"t":1,"p":0}],"s":0,"w":0,"d":3600,"f0":[]},{"0":{"r":[{"g":4,"v":0}]},"1":{"r":[{"g":4,"v":1}]},"t":20,"b":[{"g":10,"t":1,"p":0}],"s":0,"w":0,"d":3600,"f0":[]},{"0":{"r":[{"g":15,"v":0}]},"1":{"r":[{"g":15,"v":1}]},"t":20,"b":[{"g":14,"t":1,"p":0}],"s":0,"w":0,"d":3600,"f0":[]}]}

sercitecotf, 2020-04-23:
SONOFF 4ch
2 timer water valves and 2 switches, made with haajc v0.6
Json:

{"c":{"o":0,"l":13,"i":0,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":20,"b":[{"g":0,"t":1,"p":0}],"s":0,"w":0,"d":3600,"f0":[]},{"0":{"r":[{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1}]},"t":20,"b":[{"g":9,"t":1,"p":0}],"s":0,"w":0,"d":3600,"f0":[]},{"0":{"r":[{"g":15,"v":0}]},"1":{"r":[{"g":15,"v":1}]},"t":1,"b":[{"g":14,"t":1}],"s":0},{"0":{"r":[{"g":4,"v":0}]},"1":{"r":[{"g":4,"v":1}]},"t":1,"b":[{"g":10,"t":1}],"s":0}]}

pilot1981, 2020-04-25:
- SONOFF BASIC R3 RF
- 1 lock mechanism and dummy switch to enter in setup mode
- before flash HAA, I had pairing with SONOFF DW1 that send signal when it open to SONOFF BASIC R3 RF (I use RF signal to work in long outdoor distance)
- my scenario is to domotize traditional mailbox of my home: it send notification when you receive a new mail in your HW mailbox!
- If you are interested I can share also some detail on 433 MHz antenna modification! No other HW modification are made on SONOFF BASIC R3 RF and SONOFF DW1

JSON:
{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"t":4,"s":5,"i":1,"1":{"r":[{"g":12,"v":0}]},"0":{"r":[{"g":12,"v":1,"i":0.5}]},"b":[{"g":0,"p":1,"t":1}]},{"i":0.5,"1":{"s":[{"a":1}]}}]}

[image]

TriderG75, 2020-04-29:
- Electrodragon ESP LED Strip Board, Mosfet Drive
https://www.electrodragon.com/product/esp-led-strip-board/   (Tested)

- Json for managing a led strip and a dummy switch that allows you to enter in setup mode by pressing it once.

- Json:

 RGB
 {"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":15,"g":13,"v":12,"b":[{"g":0}]},{"i":0.5,"1":{"s": [{"a":1}]}}]}

 RGBW
 {"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":15,"g":13,"v":12,"w":14,"b":[{"g":0}]},{"i":0.5,"1":     {"s":[{"a":1}]}}]}

[image]

tonysprenk, 2020-04-30:
[Zemismart Curtain Motor and Rail](https://a.aliexpress.com/_BUv4oO) 

Inexpensive automatic curtain solution. It uses a WiFi usb dongle. Inside the usb dongle is an esp8266 which can be flashed with HAA using [Tuya-Convert](https://github.com/ct-Open-Source/tuya-convert) . The esp8266 chip communicates with a secondary mcu inside the motor housing. With the below json the curtain motor can be controlled through the home app however when controlling manually or with the provided RF remote the status is not updated in the home app as this is not (yet) supported by HAA. 

JSON:

```

{
	"c": {
		"r": [{
			"n": 2,
			"s": 9600
		}]
	},

	"a": [{
		"t": 45,
                 "o": 13,
		"c": 13,
                 "0": {
			"u": [{
				"v": "55aa00060005660400010075"
			}]

		},

		"1": {
			"u": [{
				"v": "55aa00060005660400010176"
			}]

		},
                 
                 "2": {
			"u": [{
				"v": "55aa00060005660400010277"
			}]

		},

		"y1": [{
			"v": 1,
			"1": {
				"u": [{
					"v": "55aa00060008650200040000000179"
				}]
			}
		}, {
			"v": 5,
			"1": {
				"u": [{
					"v": "55aa0006000865020004000000057d"
				}]
			}
		}, {
			"v": 10,
			"1": {
				"u": [{
					"v": "55aa00060008650200040000000a82"
				}]
			}
		}, {
			"v": 15,
			"1": {
				"u": [{
					"v": "55aa00060008650200040000000f87"
				}]
			}
		}, {
			"v": 20,
			"1": {
				"u": [{
					"v": "55aa0006000865020004000000148c"
				}]
			}
		}, {
			"v": 25,
			"1": {
				"u": [{
					"v": "55aa00060008650200040000001991"
				}]
			}
		}, {
			"v": 30,
			"1": {
				"u": [{
					"v": "55aa00060008650200040000001e96"
				}]
			}
		}, {
			"v": 35,
			"1": {
				"u": [{
					"v": "55aa0006000865020004000000239b"
				}]
			}
		}, {
			"v": 40,
			"1": {
				"u": [{
					"v": "55aa000600086502000400000028a0"
				}]
			}
		}, {
			"v": 45,
			"1": {
				"u": [{
					"v": "55aa00060008650200040000002da5"
				}]
			}
		}, {
			"v": 50,
			"1": {
				"u": [{
					"v": "55aa000600086502000400000032aa"
				}]
			}
		}, {
			"v": 55,
			"1": {
				"u": [{
					"v": "55aa000600086502000400000037af"
				}]
			}
		}, {
			"v": 60,
			"1": {
				"u": [{
					"v": "55aa00060008650200040000003cb4"
				}]
			}
		}, {
			"v": 65,
			"1": {
				"u": [{
					"v": "55aa000600086502000400000041b9"
				}]
			}
		}, {
			"v": 70,
			"1": {
				"u": [{
					"v": "55aa000600086502000400000046be"
				}]
			}
		}, {
			"v": 75,
			"1": {
				"u": [{
					"v": "55aa00060008650200040000004bc3"
				}]
			}
		}, {
			"v": 80,
			"1": {
				"u": [{
					"v": "55aa000600086502000400000050c8"
				}]
			}
		}, {
			"v": 85,
			"1": {
				"u": [{
					"v": "55aa000600086502000400000055cd"
				}]
			}
		}, {
			"v": 90,
			"1": {
				"u": [{
					"v": "55aa00060008650200040000005ad2"
				}]
			}
		}, {
			"v": 95,
			"1": {
				"u": [{
					"v": "55aa00060008650200040000005fd7"
				}]
			}
		}, {
			"v": 100,
			"1": {
				"u": [{
					"v": "55aa000600086502000400000064dc"
				}]
			}

		}
]
	},
{
		"1": {
			"s": [{
				"a": 1
			}]
		},
		"i": 1
	}]

}
```

aleroot, 2020-05-07:
Sonoff Basic R2 with Switch connected to GPIO 3 (RX) and GND :

`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":3},{"g":3,"t":0}]}]}`

Andre000Dias, 2020-05-07:
json for garage door? sonoff basic r2 (dry contact) HAA homekit pls

KrzysiekRadek, 2020-05-09:
I need a sprinkler script entry for Sonoff basic ... can you help me?

fribor, 2020-05-11:
Gosund SP1 v23 Outlet, obviously without the smart metering feature. 

{"c":{"l":2,"b":[{"g":3,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":14},{"g":1,"v":1}]},"1":{"r":[{"g":14,"v":1},{"g":1}]},"b":[{"g":3}]}]}

L4RGER, 2020-05-13:
Zemismart 2 Gang/ Connect Smarthome 2 Gang
KS-811 2
Tested and working

wayn3, 2020-05-13:
Lucci Fan Controller (AU Certified)
SKU: 299041
Tested and working flashed OTA with Tuya Convert
3 Speed fan control + non-dimable light control

{
    "c": {
        "r": [{
            "n": 2,
            "s": 9600
        }],
		"n":"Ceiling Fan"
    },
    "a": [{
        "t": 65,
		"e": 3,
		"0":{"u":[{"v":"55AA0006000501010001000D"}]},
        "y0": [
			{"v":1, "0":{"u":[{"v":"55AA00060005020400010011"}]}},
			{"v":2, "0":{"u":[{"v":"55AA00060005020400010112"}]}},
			{"v":3, "0":{"u":[{"v":"55AA00060005020400010213"}]}}
		]
	},{
		"t": 1,
		"0":{"u":[{"v":"55AA00060005660100010072"}]},
		"1":{"u":[{"v":"55AA00060005660100010173"}]}
    }
	]
}

MikeDProducer, 2020-05-13:
Merkury MI-WW105-199W Plug.

With this configuration the LED on the switch does not stay on when the device is powered on or off. It does light up with sending or receiving wireless signals. This could be modified if you prefer to see the LED lit up all the time.

[image]

{
  "c": { "l": 5, "b": [{ "g": 13, "t": 5 }] },
  "a": [{
    "t": 2,
    "s": 5,
    "0": { "r": [{ "g": 12, "v": 0 }] },
    "1": { "r": [{ "g": 12, "v": 1 }] },
    "b": [{ "g": 13 }]
  }]
}

wayn3, 2020-05-15:
Mirabella Genio Wi-Fi SES LED Candle #1002745 (E14 Small Edison Screw Cap)
https://mirabellagenio.net.au/ses-led-candle

{
   "c":{
	  "n":"Mirabella E14 Light"
   },
   "a":[
      {
         "t":30,
         "v":14,
         "w":12,
         "s":0
      }
   ]
}

L4RGER, 2020-05-15:
[image]
Zemismart 3 Gang/ Connect Smarthome 3Gang AU
KS-811 3/ CSH-SWITCH3
Tested and working flashed ota w/ Tuya-convert
- Credit to Michael C

{
	"c": {
		"b": [{
			"g": 4,
			"t": 5
		}]
	},
	"a": [{
		"0": {
			"r": [{
				"g": 13,
				"v": 0
			}]
		},
		"1": {
			"r": [{
				"g": 13,
				"v": 1
			}]
		},
		"t": 1,
		"b": [{
			"g": 16,
			"t": 1
		}],
		"s": 1,
		"f0": [],
		"f1": []
	}, {
		"0": {
			"r": [{
				"g": 12,
				"v": 0
			}]
		},
		"1": {
			"r": [{
				"g": 12,
				"v": 1
			}]
		},
		"t": 1,
		"b": [{
			"g": 5,
			"t": 1
		}],
		"s": 0
	}, {
		"0": {
			"r": [{
				"g": 14,
				"v": 0
			}]
		},
		"1": {
			"r": [{
				"g": 14,
				"v": 1
			}]
		},
		"t": 1,
		"b": [{
			"g": 4,
			"t": 1
		}],
		"s": 0
	}]
}

TriderG75, 2020-05-22:
- Sonoff Basic

- This json is for managing a garage door that also has a timer switch and a dummy switch that allows you to enter in setup mode by pressing it once. By tapping the garage icon in Homekit App, it opens the garage door (opening time 15sec.). After 20 seconds of the garage door being  fully open, it closes itself (closing time 15sec.)
(Tested with Lubanski on Discord)

- Json

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"s":5,"d":15,"0":{"r":[{"g":12,"v":1,"i":1}]},"1":{"r":[{"g":12,"v":1,"i":1}],"m":[{"g":2,"v":1},{"g":2,"v":-1}]},"5":{"m":[{"g":2,"v":0}]}},{"t":1,"xa":0,"d":35,"0":{"m":[{"g":1,"v":1}]}},{"i":0.5,"1":{"s":[{"a":1}]}}]}

capitantrueno, 2020-05-27:
> Teckin SP22
> Smart Plug
> 
> `{"c":{"l":3,"b":[{"g":1,"t":5}]},"a":[{"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1}]},"t":2,"b":[{"g":1}]}]}`

Would it be possible for the blue LED (GPIO 13) to remain on to indicate that the plug is in operation? I tried adding an action to the list, but it didn't work. 

This is what I tried:
`{"c":{"l":3,"b":[{"g":1,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1},{"g":13,"v":1}]},"b":[{"g":1}]}]}`

Thank you all for your great work!

Best regards!

wayn3, 2020-06-09:
DETA Smart 18W 1900lm T8 Tube (AU Certified)
SKU: DET018HA
Tested and working flashed OTA with Tuya Convert
dimable single colour tube light to retrofit fluorescent tube fittings
https://www.bunnings.com.au/deta-smart-18w-1900lm-t8-tube-with-grid-connect_p0111511
[image]

```
{
	"c":{"n":"Deta T8 Tube Light"},
	"a":[{
		"t":30,
		"w":14,
		"s":1
	}]
}
```

hillaliy, 2020-06-10:
- LoraTap Fan / Light switch
[image]

- Json:
{"c":{"r":[{"n":0,"s":9600}]},"a":[{"t":65,"e":3,"0":{"u":[{"v":"55AA0006000501010001000D"}]},"1":{"u":[{"v":"55AA0006000501010001010E"}]},"y0": [{"v":1,"0":{"u":[{"v":"55AA00060005030400010012"}]}},{"v":2, "0":{"u":[{"v":"55AA00060005030400010113"}]}},{"v":3,"0":{"u":[{"v":"55AA00060005030400010214"}]}}]},{"t": 1,"0":{"u":[{"v":"55AA00060005090100010015"}]},"1":{"u":[{"v":"55AA00060005090100010116"}]}}]}

[124 more replies]

closed 2021-08-13 by RavenSystem