RavenSystem/esp-homekit-devices #690: Error in initial Setup, can't reflash

brymonster, 2019-12-24:
Hello all. 
Successfully flash another Magic Home RGBW device identical to the [previous.](https://github.com/RavenSystem/esp-homekit-devices/issues/654#issuecomment-564269037) Unfortunately, when connecting to the initial Setup page I was only shown the Network List. No OTA repository or OTA binary file fields.
I selected my network, entered password and joined. Now the device is on the network, but I have no way to enter Setup Mode again. 
The install guide mentions "If installation fails, you must start all process again from step 2."
I have tried reflashing but keep getting the `Connecting........_____....._____` 
Could something be preventing refreshing?

Previously I was able to get the device into flash mode by plugging in to FDTI while jumping the io0 to ⏚ for 3-5 seconds. Now im trying lots of variations, nothing is working so far. 

Ideas?

brymonster, 2019-12-25:
have been able to test on another computer. Same result:
```
$ esptool.py -p /dev/tty.usbserial-A50285BI erase_flash
esptool.py v2.8
Serial port /dev/tty.usbserial-A50285BI
Connecting........_____....._____.....
```

GPL71, 2019-12-25:
Try swap TX and RX on FTDI

brymonster, 2019-12-26:
No joy 
Would something on a bad flash be preventing a device entering Flashing Mode?
How long to you hold the io0 to ⏚?

brymonster, 2019-12-26:
Sorry to bother friends. My FTDI had a loose solder on the ⏚.
Solved!

closed 2019-12-26 by brymonster