RavenSystem/esp-homekit-devices #691: Request for  HHA for Sonoff Dual Window Covering, with 2 external switches ,open and close time 25 sec.

sblitzer, 2019-12-25:
Hello,

For now I have using RC 1.2 for Window Covering and works perfectlly.
I looking for HAA template for Sonoff Dual - Window Covering, including 2 external switches ,open and close time 26 sec.
Can anyone create this template for me please? 
I would greatly appreciate it

Dirk23, 2019-12-25:
Hi. I have Sonoff Dual R2 and i am looking forward to use the same. But i am not able to get to the point where i can enter Setup. Can you tell me how you did it?

sblitzer, 2019-12-26:
> Hi. I have Sonoff Dual R2 and i am looking forward to use the same. But i am not able to get to the point where i can enter Setup. Can you tell me how you did it?

Hi Dirk - I posted a request - I still not update my Window Covering to HAA - right now using RC V1.2
But if you flashed your device already with HAA click this link ( https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode ) to learn how to enter setup mode.
Good luck

Dirk23, 2019-12-26:
> Hi Dirk - I posted a request - I still not update my Window Covering to HAA - right now using RC V1.2
> But if you flashed your device already with HAA click this link ( [...]

Of course I tried to set my Sonoff Dual R2 into setup mode after reading the instructions in the wiki. I even have a serial logging on to see if anything happens, but neither the eight times you press the S1 (built-in button that can be reached from outside the cover), nor the Button0 or Button1 switches do anything, not to mention that a website is accessible under the IP of the device.

sblitzer, 2019-12-26:
> Of course I tried to set my Sonoff Dual R2 into setup mode after reading the instructions in the wiki. I even have a serial logging on to see if anything happens, but neither the eight times you [...]

The truth is I was able to enter setup mode with eight time press on the built in button but with Sonoff basic and it's OK

kafes33, 2019-12-27:
This Is it i think:
{"c":{"l":13,"b":[{"g":10,"t":5}]},"a":[{"t":45,"o":26,"c":26,"f":0,"0":{"r":[{"g":12,"v":1},{"g":5}]},"1":{"r":[{"g":12},{"g":5,"v":1}]},"2":{"r":[{"g":12},{"g":5}]},"3":{"r":[{"g":12,"i":1},{"g":5}]},"4":{"r":[{"g":12},{"g":5,"i":1}]},"f0":[{"g":0}],"f1":[{"g":9}]}]}

And my question Is:
Is it possible Add function to buttons f0 and f1, for stop / closing / opening / blinds?

like this:
press f0 butron / blinds are closing / then press button f0 again and blinds stop closing?
press f1 / blinds are opening / then press button f1 again and blinds stop opening?

thank you very much for your answer, and for your amazing  work!.)
have a nice evening.

sblitzer, 2019-12-27:
WoW - works like a charm
Thank you so much

closed 2019-12-27 by sblitzer