RavenSystem/esp-homekit-devices #692: Flashing with esptool throws an Error

Dirk23, 2019-12-25:
Hi,

i tried to flash a Sonoff Dual R2 with HAA. In your Wiki is written:

`esptool.py -p /dev/<your_ESPPort> --baud 115200 write_flash -fs 1MB -fmdout -ff 40m \
0x0 rboot.bin 0x1000 blank_config.bin 0x2000 otaboot.bin` but the option fmdout is unkown:

[image]

I changed it to `--flash_mode dout` but i don't know if this correct. Anyways, after flashing the Sonoff, i could setup wifi, but i can't enter Setup Mode. I press the OnBoard Button 8 times, and i tried it with the 2 external Switch Pins (button0 and button1) but there is no Webpage reachable after pressing these Buttons 8 times.

What am i missing?

Regards

Dirk

GPL71, 2019-12-25:
Simply enlarge (horizontal) the terminal window ... you have typed command in 2 rows.

Dirk23, 2019-12-25:
Nope, that is not the problem, there was no linebreak or anything:
[image]
I just realized that `fmdout` is wrong and that it needs a blank space between -fm and dout like `-fm dout`
I think you need to correct your wiki.

closed 2019-12-25 by Dirk23