RavenSystem/esp-homekit-devices #693: Help with Blitzwolf BW-LT11 led strip

peros550, 2019-12-26:
@RavenSystem Hi! 

I flashed a Blitzwolf Led strip (BW-LT11) using the tuya-convert method. If possible, I'd like to ask for somesome help with the JSON. 

Here is the JSON i used: 
`{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":4,"g":12,"v":5,"w":14,"b":[{"g":0}]}]} `

Because it's very difficult to describe the issue, I recorded a short video. And here is what happens when changing colors: 
https://www.youtube.com/watch?v=xGukbIiIf-k

The blue does not appear at all. Any ideas?

RavenSystem, 2019-12-26:
GPIOs are wrong. Swap GPIO 5 and GPIO 14.

https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices-(HTTP-firmware)#lights

peros550, 2019-12-26:
Thanks! It worked perfectly fine!!!!! Color reproduction is very nice. 

I would like to solder some cables (GPIO2 looks free) and add a motion sensor, hence the Led Strip would turn on automatically to white upon movement. Do you think this is possible with current version?

Thanks again!

RavenSystem, 2019-12-26:
You can add a motion sensor, and you can turn on light when motion is detected (without using HomeKit automations), but only turn on/off light; you can not set a color: it will use the last selected color.

peros550, 2019-12-26:
Thanks! Yes figured out how to  I add a motion sensor but haven't so far figured out how to add the automation internally within the HAA. 

`{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":4,"g":12,"v":14,"w":5,"b":[{"g":0}],"s":0,"p":1024,"d":1,"e":10,"fr":1,"fg":1,"fv":1},{"1":{"r":[{"g":0,"v":0}]},"t":12,"s":0,"f0":[],"f1":[{"g":13,"t":1}],"i":4}]}`

RavenSystem, 2019-12-26:
This JSON will turn on light when motion is detected, but it will not turn off light when motion is not detected. I removed all unnecessary keys because they use default values, and declaring they can create memory issues:

`{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":4,"g":12,"v":14,"w":5,"b":[{"g":0}]},{"t":12,"i":4,"f1":[{"g":13}],"1":{"m":[{"g":1,"a":1}]}}]}`

peros550, 2019-12-26:
Right now it looks that motion turns off the lights... I think we are close :)

peros550, 2019-12-26:
@RavenSystem this is what worked for me: 

{"c":{"n":"Led-BW-LT11","b":[{"g":0,"t":5}]},"a":[{"t":30,"r":4,"g":12,"v":14,"w":5,"b":[{"g":0}]},{"1":{"m":[{"g":1,"v":1}]},"t":12,"s":0,"f0":[],"f1":[{"g":13,"t":1,"i":0,"p":1},{"g":13,"t":0,"p":1}],"i":3}]}

I think there is no way to combine it with "d":14400 command we discussed at discord to have an automatic turn off right? 

Many thanks so far! Your help is greatly appreciated!!!!!

RavenSystem, 2019-12-26:
> I think there is no way to combine it with "d":14400 command we discussed at discord to have an automatic turn off right?

You must use an additional switch with the "d" key: Motion sensor activates switch, and switch controls led strip.

peros550, 2019-12-27:
Thanks that's a good idea. I will definitely try it. I'm still trying to find my way with json though.  

One more question. I have noticed that with the above json, the motion sensor turns on the LED strip at white mode. This is nice when the LED strip is turned off, but if the LED strip is turned to a different color, setting white may feels disturbing. If there is a way to only send the command to turn ON when the strip is Off would make the perfect accessory :)

Thanks!!!!

peros550, 2019-12-31:
I had an issue yesterday in which the motion sensor got stuck to a "motion detected" state and wouldn't go back to normal. The operation seemed to be valid for both the the home app but also for the automation we built inside the esp8266. 
I had to reset power so that it could get back to normal. 
The power strip was functional through home app. 

Any ideas on how to troubleshoot similar future situation? Is there any way to remotely get logs on the device for further troubleshooting?

RavenSystem, 2019-12-31:
@peros550 JSON?

peros550, 2019-12-31:
This is it: 
`{"c":{"n":"Led-BW-LT11","b":[{"g":0,"t":5}]},"a":[{"t":30,"r":4,"g":12,"v":14,"w":5,"b":[{"g":0}]},{"1":{"m":[{"g":1,"v":1}]},"t":12,"s":0,"f0":[],"f1":[{"g":13,"t":1,"i":0,"p":1},{"g":13,"t":0,"p":1}],"i":3}]}`

RavenSystem, 2019-12-31:
`{"g":13,"t":1,"i":0,"p":1},{"g":13,"t":0,"p":1}` -> here you are telling motion sensor that motion is detected when gpio 13 is high and low. That is wrong.

I don't know if your motion sensor sets pin to high or low when motion is detected. Try this with pin to high when motion is detected:
`{"c":{"n":"Led-BW-LT11","b":[{"g":0,"t":5}]},"a":[{"t":30,"r":4,"g":12,"v":14,"w":5,"b":[{"g":0}]},{"1":{"m":[{"g":1,"v":1}]},"t":12,"f1":[{"g":13}],"i":3}]}`

peros550, 2019-12-31:
Thanks Jose! It just works! But is there any way to get remote logs for troubleshooting purposes?

RavenSystem, 2020-01-02:
> Thanks Jose! It just works! But is there any way to get remote logs for troubleshooting purposes?

No.

peros550, 2020-01-02:
No worries! 
Thank you for all the support!!!!

peros550, 2020-01-02:
I would like to report a strange situation with the above RGBW accessory in version 1.1. 

Today, the accessory was uncontrollable from home app (red no response) while the light was actually turned on. I pressed the hardware button a few times and the accessory was available again in home app. While I was pressing the hardware button a few times, I was able to see the accessory change status in home app, but in reality it never changed status. It remained constantly turned on. 
From a network perspective, I was able to ping the accessory successfully while in red no response.

RavenSystem, 2020-01-02:
It is fixed some minutes ago.

peros550, 2020-01-02:
Ok Jose! Good to know! Thanks again! 
One more question in regards to the below json: 
`{"c":{"n":"Led-BW-LT11","b":[{"g":0,"t":5}]},"a":[{"t":30,"r":4,"g":12,"v":14,"w":5,"b":[{"g":0}]},{"1":{"m":[{"g":1,"v":1}]},"t":12,"f1":[{"g":13}],"i":3}]}`

This custom json has a motion sensor turn on the Led strip any time is senses motion. But when the led is already turned on to another color, it resets the color to white. Do you have any suggestion on could we set it to do nothing if the led is already turned on to some color? 

Many thanks!!!!!

RavenSystem, 2020-01-02:
Color must not change. Only turn on if light is off. If light is on, does nothing. I will check it.

peros550, 2020-01-02:
By the way, 1.1.1 seems to solve the out of sync status, but right after the update the device went in no response status and got right back with the press of the hardware button. In 0.9.2 seemed a lot more stable. By the way, I have another 3-4 HAA devices in 1.1 with no issues.

RavenSystem, 2020-01-02:
Are you using Asus Merlin?

peros550, 2020-01-02:
In my home yes, but at the moment we are at my father's in law house and there is a TP-Link router.

RavenSystem, 2020-01-02:
I have many issues with last version of Asus Merlin 384.14, but yesterday a new minor update was released 384.14_2 and now all seem to be right.

Before I had a TP-Link router, and sometimes I got that issue when many accessories were connected (for a TP-Link many is 6-8 accessories). Reboot your TP-Link to flush mDNS cache.

RavenSystem, 2020-01-02:
Other thing. I tested motion sensor and light using your JSON, and I always get the actual color and brightness, not white.

peros550, 2020-01-02:
In my home I have still 384.13 and all HAA run fine :)

You are right about the color. I'm sorry for the false alarm :( and for wasting your time. 

Do you see any issues with GPIO13 ? I would need to solder a new cable directly to the esp8266, I can do it tomorrow. 

BTW, no response issue is still present after reboot of ATV and TP-Link. All devices are OK except the BlitzWolf Led Strip. The Led strip come alive again if I press the hardware button.

RavenSystem, 2020-01-02:
> In my home I have still 384.13 and all HAA run fine :)

Keep it. 384.14 is a mess. I hope 384.14_2 is stable.

> You are right about the color. I'm sorry for the false alarm :( and for wasting your time.

Don't worry.

> Do you see any issues with GPIO13 ? I would need to solder a new cable directly to the esp8266, I can do it tomorrow.

No. I was thinking in a interference with PWM, but if color is right, never mind. Keep GPIO 13.

> BTW, no response issue is still present after reboot of ATV and TP-Link. All devices are OK except the BlitzWolf Led Strip. The Led strip come alive again if I press the hardware button.

When you press button, a notification with the new status is sent to all HK clients. It is strange, because all devices use same HomeKit functions. Unplug it for 15-20 minutes, and plug it again.

peros550, 2020-01-02:
Thanks, I will monitor the situation and let you know if any new comes to my attention! 👍

peros550, 2020-01-07:
@RavenSystem Looks good so far. 👍👍👍👍
PS: I also flashed a Blitzwolf BW-LT-21 smart bulb and used the following json: 
{"c":{"n":"Led-BW-LT21","b":[{"g":0}]},"a":[{"t":30,"r":14,"g":12,"v":5,"w":15,"b":[{"g":0}]}]}

seems to work nicely, I'm just not sure if the last part ~~"b":[{"g":0}]~~ was really needed. Let me know and I will update the examples topic with the correct json. 

One more question in regards to the colors (but of minimum priority): Noticed that when changing colors , the change does not feel as smooth as in other products (i.e. Philips Hue, Yeelight Color bulbs). For example, when turning the light off, the lights dim very smoothly, but I would expect same smoothness when changing color. Is this firmware or hardware oriented?

All the best

RavenSystem, 2020-01-08:
@peros550 

`"c":{"n":"Led-BW-LT21","b":[{"g":0}]}` -> Here `"b":[{"g":0}]` is used as mechanism to enter setup mode. In this case, a simple press on gpio-0 button.

`"a":[{"t":30,"r":14,"g":12,"v":5,"w":15,"b":[{"g":0}]}]` -> Here `"b":[{"g":0}]` is used as mechanism to turn on/off light. In this case, a simple press on gpio-0 button.

What happens? You use same button and press type for both: enter setup mode and turn on/off light. 

You can choice between:
1. Changing first to `"b":[{"g":0,"t":5}]` to enter setup mode holding button 8 seconds.
2. Keeping a simple press to enter setup mode, and remove last `"b":[{"g":0}]`.

> ... For example, when turning the light off, the lights dim very smoothly, but I would expect same smoothness when changing color. Is this firmware or hardware oriented?

Way to change brightness is the same for colors. Maybe problem is with the way to receive new target color. Try clicking on the new color instead moving your finger over the color wheel and see if there is any change with color transition.

peros550, 2020-01-08:
@RavenSystem Hi Jose, thanks for the feedback! Really much appreciated :) 

The Led-BW-LT21 is a smart bulb (E27), it has no physical button connected to GPIO 0. Is the "b":[{"g":0}] totally irrelevant? (both in the beginning and end positions) ? If I remove them, will I lose the functionality of getting into the setup mode in any way?

RavenSystem, 2020-01-08:
> I think I can safely remove the "b":[{"g":0}] in both references since there is no button on the E27 smart buld, right?

Yes.

> Entering into setup by ON-OFF is already in the firmware and does not need json configuration, right?

Yes, you can enter in setup mode triggering 8 times on/off from Home App, or using emergency setup mode.

"Triggering 8 times on/off" can be disabled if you declare and set `"z"` key to `0` in config section. More info about: https://github.com/RavenSystem/esp-homekit-devices/wiki/General-Configuration#Setup-Mode-Toggle-Count

"Emergency setup mode" can not be disabled.

peros550, 2020-01-08:
Great , thanks

closed 2020-01-08 by peros550