RavenSystem/esp-homekit-devices #696: Magic Home external toggle?

brymonster, 2019-12-26:
Hi all. Wondering if it’s possible to use one of the GPIOs to have an external toggle on a Magic Home RGBW.

mafyata, 2019-12-26:
Yes, try IO0

brymonster, 2019-12-26:
Thanks @mafyata!
How exactly do you have that set up? 
12v into io0 when the switch is on?

mafyata, 2019-12-27:
IO0 and GRN connected for a second. Like push button.

brymonster, 2019-12-27:
Ok I will have to try that when I gets home. 

Momentary grounding io0 will turn on.. and off? 
Any changes to the JSON needed?

mafyata, 2019-12-27:
You must set external button for IO0 in your json.

maccoylton, 2019-12-27:
>You must set external button for IO0 in your json.

@brymonster if you have the model WIU the the IR remote can you not use that? 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

mafyata, 2019-12-27:
I have 2 rgb devices with ir and after flashing with haa the ir is not working. Haa don’t support ir control.

brymonster, 2019-12-27:
Thanks for the help so far @mafyata. 
Here is the JSON I’m using currently: 
```
{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":12,"g":5,"v":13,"w":15,"b":[{"g":0}]}]}
```
I have been messing with the [JSON Configurator](https://glumb.github.io/haa-configurator/) But don’t have a full grasp of it. 
I think the t: value needs to be adjusted. 
Can you share the JSON your using?

mafyata, 2019-12-27:
I don’t use external button. I have a switch for setup mode.
1- {"c":{"n":"Kuhnq-Led","b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13},{"0":{"r":[]},"1":{"r":[],"s":[{"a":1}]},"t":1,"b":[],"s":0}]}

2- {"c":{"n":"Hol-Led","b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13,"fr":5,"fg":5,"fv":5},{"0":{"r":[]},"1":{"r":[],"s":[{"a":1}]},"t":1,"b":[],"s":0}]}

maccoylton, 2019-12-27:
I have firmware here that has IR support :- https://github.com/maccoylton/esp-homekit-rgbw-strip

brymonster, 2019-12-28:
I will check it out. Thanks @maccoylton. 
My current Magic Home devices don’t have IR. That why I was looking to add a connected switch. 
But maybe my future ones will. HAA also has IR on the todo list!

brymonster, 2019-12-30:
Momentary grounding io0 works with my current JSON. Thanks!

closed 2019-12-30 by brymonster