RavenSystem/esp-homekit-devices #697: json for multiple outlets in one power strip

Ghitafjorback, 2019-12-27:
I need to change my json to have more than one outlet within the same json. I have 4 outlets, plus a USB outlet. My json looks like this, and will only allow one outlet to be handled from homekit...

{"c":{"l":4,"b":[{"g":13,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":13}]}]}

pawelosin, 2019-12-27:
It’s the JSON for Power Strip SA-P402A. You must change gpio’s to the correct for your power strip. 
{"c":{"l":3,"i":1,"b":[{"g":0,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":1}]},{"t":2,"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1}]},"b":[{"g":1,"t":2}]},{"t":2,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":1,"t":3}]},{"t":2,"0":{"r":[{"g":13}]},"1":{"r":[{"g":13,"v":1}]},"b":[{"g":1,"t":4}]},{"t":2,"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":1,"t":5}]}]}

Ghitafjorback, 2019-12-28:
Thank you! that works completely. Only one thing. The USB outlets will turn ON when I turn them OFF in Homekit, and they will turn OFF when I turn them ON in Homekit.... I am not good at reading Json, but I guess it is just something that needs to be exchanged somewhere right!?

Ghitafjorback, 2019-12-28:
And one more thing. The "button"... the physical button on the power strip does not work, and does not show in Homekit. Would that maybe be possible to add in the Json?

pawelosin, 2019-12-28:
What type of power strip do you have?

Ghitafjorback, 2019-12-28:
I have a Tuya smart power strip with four regular EU outlets and 4 USB plugs (only one connector for the 4 USBs)

pawelosin, 2019-12-28:
I am not sure, but i think that the physical button in your power strip is connected to GPIO 2. Try to change the GPIO from "g":1 to "g":2 in the sections button.

Ghitafjorback, 2019-12-28:
I am very new to this. So I am not completely sure I understand. I changed as per below, and updated the json to the strip. I do not see any changes, and the physical buttons is still not working. Did I do something wrong?

{"c":{"l":3,"i":1,"b":[{"g":0,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":1**-CHANGED TO 2**}]},{"t":2,"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1}]},"b":[{"g":1**-CHANGED TO 2**,"t":2}]},{"t":2,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":1**-CHANGED TO 2**,"t":3}]},{"t":2,"0":{"r":[{"g":13}]},"1":{"r":[{"g":13,"v":1}]},"b":[{"g":1**-CHANGED TO 2**,"t":4}]},{"t":2,"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":1**-CHANGED TO 2**,"t":5}]}]}

Ghitafjorback, 2019-12-28:
Maybe I should add... there is only 1 button on the power strip.

pawelosin, 2019-12-28:
No, don’t change GPIO. It seems that configuration is correct. 
Switch1 is the physical button and it should work.
Characteristic of the button - https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#press-type

I have updated my power strip to 0.9.2 version and it still works fine. 
My json is the same as earlier:

{"c":{"l":3,"i":1,"b":[{"g":0,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":1}]},{"t":2,"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1}]},"b":[{"g":1,"t":2}]},{"t":2,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":1,"t":3}]},{"t":2,"0":{"r":[{"g":13}]},"1":{"r":[{"g":13,"v":1}]},"b":[{"g":1,"t":4}]},{"t":2,"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":1,"t":5}]}]}

What exactly is your power strip. Send me the link.

Ghitafjorback, 2019-12-28:
it is this one... TYPE-E (second picture)
[https://www.aliexpress.com/item/32921984017.html](url)

Ghitafjorback, 2019-12-28:
Since this is my first time ever flashing and working with json I have been trying a lot of stuff to see what works. latest, I was feeling very brave, and I tried to update my json to something totally different in the hope that I could somehow get the button to work.... now I just feel stupid.... And very close to crying actually....

I changed my json to 
{"c": {"o": 0,"i": 1,"l": 3,"m": 5},"a": [{"0": {"r": [{"g": 5,"v": 0}]},"1": {"r": [{"g": 5,"v": 0}]},"t": 2,"b": [],"s": 1,"f0": [{"g": 5,"t": 1,"i": 0}],"f1": [{"g": 5,"t": 1,"i": 0,"p": 0}]}]}

What I expected it to do did NOT work... surprise... well I have try things to learn I guess.... BUT the worst part is, now I cannot enter the setup-mode again. To enter the setup-mode I need to toggle one of the outlets on/off several times (I did that from Homekit before), but since my new json doesn't allow me to toggle any of the outlets, then I cannot enter setup-mode to change the json.

pawelosin, 2019-12-28:
You can to reflash it.

Ghitafjorback, 2019-12-28:
In order to flash it, I still need to access the setup mode.

mafyata, 2019-12-28:
First start after reflash will enter the device in setup mode automatically. Set an button for setup mode. I created a room for setup mode buttons.
[image]

Ghitafjorback, 2019-12-28:
That is really cool with the setup mode button! How did you do that? Is it part of the json?
I know that the setup mode will be entered automatically after flashing, but in order to reflash my device, I need to enter setup mode. I cannot enter setup mode, as the physical button will not react, nor can I toggle any of the outlets from HomeKit, as my json isn’t allowing that.
I can see the device on my network, I just cannot enter it, since I cannot get it to enter the setup mode. I tried putting the device’s up address in the browser, but again, since the device is not in setup mode, nothing. 😫😫

Olepir, 2020-01-03:
I found correct GPIO for this strip
https://github.com/arendst/Tasmota/pull/4590
So JSON can look like this. Is someone can confirm? 
I also putted double press for a cyclic input for each relay.

{"c":{"l":1,"i":1,"b":[{"g":3,"t":5}]},"a":[{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"t":2,"b":[{"g":3,"t":2}]},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"t":2,"b":[{"g":3,"t":2}]},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"t":2,"b":[{"g":3,"t":2}]},{"0":{"r":[{"g":13}]},"1":{"r":[{"g":13,"v":1}]},"t":2,"b":[{"g":3,"t":2}]},{"0":{"r":[{"g":14}]},"1":{"r":[{"g":14,"v":1}]},"t":2,"b":[{"g":3,"t":2}]}]}

closed 2020-01-10 by RavenSystem