RavenSystem/esp-homekit-devices #699: HAA for Sonoff Dual Window Covering

kafes33, 2019-12-28:
Hi,
This Is example code for Window covering using sonoff dual - it works great.

{"c":{"l":13,"b":[{"g":10,"t":5}]},"a":[{"t":45,"o":26,"c":26,"f":0,"0":{"r":[{"g":12,"v":1},{"g":5}]},"1":{"r":[{"g":12},{"g":5,"v":1}]},"2":{"r":[{"g":12},{"g":5}]},"3":{"r":[{"g":12,"i":1},{"g":5}]},"4":{"r":[{"g":12},{"g":5,"i":1}]},"f0":[{"g":0}],"f1":[{"g":9}]}]}

And my question Is:
Is it possible Add function to buttons f0 and f1, for stop / closing / opening / blinds?

like this:
press f0 butron / blinds are closing / then press button f0 again and blinds stop closing?
press f1 / blinds are opening / then press button f1 again and blinds stop opening?

thank you very much for your answer, and for your amazing work!.)
have a nice evening.

RavenSystem, 2019-12-30:
Updated: https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#sonoff-dual-window-covering-driver-with-external-push-buttons-status-led-and-built-in-setup-button-opening-time-18-seconds-closing-time-14-seconds-non-lineal-correction-70

closed 2019-12-30 by RavenSystem

kafes33, 2020-01-01:
Ok. it is a simple swap "fx", without adding next - now i get it:) Thank you and Happy new year!