RavenSystem/esp-homekit-devices #701: HAA Sonofff basic - how to configure The Initial State when device enters on boot

sblitzer, 2019-12-30:
Hello

For Sonofff basic i used the examples in the WIKI:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}

my question is : how do i configure Initial State of the switch when device enters on boot ?
I know that Initial state is defined by the "s" object contained within the accessory object.
For exp. i need it to be ON so the State is "1"
But how it reflected in the code?

Thanks

mafyata, 2019-12-30:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"s":1,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}

sblitzer, 2019-12-30:
mafyata - thanks for your replay - will try it today.
Have a nice day :)

sblitzer, 2019-12-30:
BTW - can you please also help with the same issue but on Sonoff dual?
I'm using this:

{"c": {"l": 13,"b": [{"g": 10}]},"a": [{"0": {"r": [{"g": 5}]},"1": {"r": [{"g": 5,"v": 1}]},"b": [{"g": 0}]},{"0": {"r": [{"g": 12}]},"1": {"r": [{"g": 12,"v": 1}]},"b": [{"g": 9}]}]}

Many thanks

mafyata, 2019-12-30:
{"c": {"l": 13,"b": [{"g": 10}]},"a": [{"s":1,"0": {"r": [{"g": 5}]},"1": {"r": [{"g": 5,"v": 1}]},"b": [{"g": 0}]},{"0": {"r": [{"g": 12}]},"1": {"r": [{"g": 12,"v": 1}]},"b": [{"g": 9}]}]}

sblitzer, 2019-12-30:
U the best !!
Many thanks

closed 2019-12-30 by RavenSystem