RavenSystem/esp-homekit-devices #702: Cant complete setup, no web appears on LCM AP

jvhtec, 2019-12-30:
Hi, just flashed a generic ESP breaker with LCM via Tasmota firmware upgrade with a file provided here on this forum. Now my device is broadcasting LCM AP but doesnt show the web configuration page on connecting. Tried from PC (Chrome, Firefox), MacOS (Safari) and iOS to no avail. Any ideas?

Rocket200, 2019-12-31:
same problem here

Dirk23, 2020-01-01:
I have the same problem with my Sonoff Dual R2. What helped me was rebooting the device. Only at the 2nd start after flashing the setup page was displayed. It seems to me that the device does not get an Ip address from the HAA Access Point.

peros550, 2020-01-01:
Same problem was found, one or multiple reboots definitely helped. AP does not assign IP when is stuck.

Rocket200, 2020-01-01:
I can now connect to the LCM Settings always (sometimes need unplug the device)
i can write everything inside wifipassword,OTA, but nothing happens after i Click Join Button (i Try Safari,Mozilla,Brave)

dbartelmus, 2020-01-04:
Whole process is not working well, I’m always  struggling to install haa in sonoff - need to wait for fix

Rocket200, 2020-01-05:
It need sometimes a reboot then get connected! (Flash with esptool)

The best way is to connect Serial and look in the log what happened

Rocket200, 2020-01-05:
> It need sometimes a reboot then get connected! (Flash with esptool)
> 
> The best way is to connect Serial and look in the log what happened

closed 2020-01-10 by RavenSystem