RavenSystem/esp-homekit-devices #704: separate accessory for LED light on smart plug

tonysprenk, 2019-12-30:
Can someone help me tweak my template so I can get I separate switch in the home app for the LED on my smart plug? 

this is the template so far:

{"c":{"b":[{"g":13,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":5},{"g":12}]},"1":{"r":[{"g":5,"v":1},{"g":12,"v":1}]},"b":[{"g":13}]}]}

Thanks a lot!!

peros550, 2019-12-31:
So you want to separate the LED function from the operation of your smart plug?

tonysprenk, 2019-12-31:
Yes, I have two smart plugs in the bedroom and the led light is quite bright when the room is dark. Would be nice to be able to turn it off separately from the relay.

peros550, 2019-12-31:
So you would still need the Led light to follow the plug operation, but at times you would like this to be completely turned off despite the plug's relay status. I'm don't know how this conditional behavior could be implemented. But maybe someone more experienced could provide more feedback...

RavenSystem, 2019-12-31:
Relay: GPIO 5
LED: GPIO 12

`{"c":{"b":[{"g":13,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":5}],"m":[{"g":2}]},"1":{"r":[{"g":5,"v":1}],"m":[{"g":2,"v":1}]},"b":[{"g":13}]},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]}}]}`

tonysprenk, 2019-12-31:
Amazing thank you!

closed 2019-12-31 by tonysprenk