RavenSystem/esp-homekit-devices #706: HAA ! Invalid JSON

fanlessfan, 2019-12-31:
try to flash sonoff basic (Sonoff RF R2 POWER in PCB) with latest v0.92 and got error below
`
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2019.12.31 13:47:42 =~=~=~=~=~=~=~=~=~=~=~=
rld|l|$c|{clco'lnobxl;l{lpndco|lcgod$`dgso$p'sco<8ogl`od`gso`;g {$bog8|clc'x`c#cog'cncp~gloggbdlp# nollgd nocdd${$bog$no'no$#og$no'`;lgn|'cll|o8oglgo8'|cpcdl`pp_task_hdl : 3ffefe48, prio:14, stack:512
pm_task_hdl : 3ffef8b0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3fe0, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Dec 22 2019 17:28:27
phy ver: 273, pp ver: 8.3

HAA ! Invalid JSON
mode : sta(84:0d:8e:5b:10:96) + softAP(86:0d:8e:5b:10:96)
add if0
add if1
bcn 100
scandone
no 8DB0839D found, reconnect after 1s
del if0
bcn 0
del if1
usl
sul 0 0

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
WAl<$'x'odg'nxo~;g|#xcld`pp_task_hdl : 3ffefe48, prio:14, stack:512
pm_task_hdl : 3ffef8b0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3fe0, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Dec 22 2019 17:28:27
phy ver: 273, pp ver: 8.3

HAA ! Invalid JSON
mode : sta(84:0d:8e:5b:10:96) + softAP(86:0d:8e:5b:10:96)
add if0
add if1
bcn 100
scandone
no 8DB0839D found, reconnect after 1s
del if0
bcn 0
del if1
usl
sul 0 0

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
WAl<$'p'odo'nxo?{|#xcld`pp_task_hdl : 3ffefe48, prio:14, stack:512
pm_task_hdl : 3ffef8b0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3fe0, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Dec 22 2019 17:28:27
phy ver: 273, pp ver: 8.3

HAA ! Invalid JSON
mode : sta(84:0d:8e:5b:10:96) + softAP(86:0d:8e:5b:10:96)
add if0
add if1
bcn 100
scandone
no 8DB0839D found, reconnect after 1s
del if0
bcn 0
del if1
usl
sul 0 0

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
WAd|ldgpgndogpn{co|cxcld`pp_task_hdl : 3ffefe48, prio:14, stack:512
pm_task_hdl : 3ffef8b0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3fe0, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Dec 22 2019 17:28:27
phy ver: 273, pp ver: 8.3

HAA ! Invalid JSON
mode : sta(84:0d:8e:5b:10:96) + softAP(86:0d:8e:5b:10:96)
add if0
add if1
bcn 100
scandone
no 8DB0839D found, reconnect after 1s
del if0
bcn 0
del if1
usl
sul 0 0

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
WA$|cgpgodog'xn?{#g|cxcld`pp_task_hdl : 3ffefe48, prio:14, stack:512
pm_task_hdl : 3ffef8b0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3fe0, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Dec 22 2019 17:28:27
phy ver: 273, pp ver: 8.3

HAA ! Invalid JSON
mode : sta(84:0d:8e:5b:10:96) + softAP(86:0d:8e:5b:10:96)
add if0
add if1
bcn 100
scandone
no 8DB0839D found, reconnect after 1s

`

fanlessfan, 2019-12-31:
flash command is 
`
esptool -p /dev/ttyS7 --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 main.bin

`

GPL71, 2019-12-31:
Flash using the otaboot.bin instead main.... after check what will happen

RavenSystem, 2019-12-31:
Did you erase flash before?

fanlessfan, 2019-12-31:
@RavenSystem @GPL71 thank you all so much. it works after erase the flash. happy new year!

closed 2019-12-31 by fanlessfan