RavenSystem/esp-homekit-devices #707: Garage Door shows as Opening in Home when powered on with door closed

MrCmonster, 2019-12-31:
Great work on this!
I'm using a Sonoff SV for a Garage Door opener with HAA 1.0.0. I'm getting odd behavior when the device boots up.

**Prerequisite:**
The device has both an open and a close sensor correctly configured.

**Steps to reproduce:**

1. Close the door
2. Once the door is fully closed, simulate a power outage by unplugging the Sonoff and plugging it back in
3. Go into the Home app and check the door status

**Expected result:**
The Sonoff should read the door sensor on startup and see that the door is closed.

**Actual result:**
The Sonoff shows that the door is Opening.
[image]

Note:
If the door is fully open when the Sonoff is power cycled, then it shows correctly that the door is open. 

Thanks!

RavenSystem, 2020-01-01:
I will check it.

pilot1981, 2020-01-01:
@MrCmonster what happen if there is power outage or a security stop when door is opening or closing? Your Sto arrivando! Able to start again relays to open/close garage door?

RavenSystem, 2020-01-01:
@MrCmonster Bug confirmed and fixed in new version 1.1.0. Try it and comment, please.

MrCmonster, 2020-01-02:
It’s fixed in 1.1.0 
Thank you very much for all of your hard work!

closed 2020-01-02 by MrCmonster