RavenSystem/esp-homekit-devices #710: How to enter Setupmode from HomeKit for Windowcover

Dirk23, 2020-01-01:
Hello,

i would like to be able to put my Sonoff Dual R2 as a roller blind in the stetupmode, if they are installed in the wall. Unfortunately it is not possible to reach the setup mode by pressing "roll up" "roll down" in Homekit. Is there a possibility? I don't want to have to unscrew 20 devices from the installation box to make an OTA.

Maybe it would be possible to do a one-time OTA by pressing a key combination, that would be great, because that would save you from having to search for the lease in DHCP to get to the setup website.

peros550, 2020-01-01:
A tool to centrally manage HAA devices makes more and more sense.

pilot1981, 2020-01-01:
I agree with you! This is more useful on HAA than RC. In this architecture we have different state of devices:
1) after flash, AP MODE
2) IP address in first SETUP MODE without JSON
3) IP address in SETUP MODE with incorrect JSON
4) configured device but without HOMEKIT connection
5) configured device but with HOMEKIT connection
6) enter again in SETUP MODE (In my case very frustrating)
7) OTA UPDATE
8) ...

I don’t included in list the complexity to create correct JSON...

In HAA we have many feature/more configuration flexibility but this introduced more complexity in implement/maintain all devices.

I’m sure @RavenSystem already know it: I trust him! I think him can find soon a solution for it!

pilot1981, 2020-01-02:
I can suggest to create something similar to this tool of TASMOTA in HOME ASSISTANT:

https://github.com/arendst/Tasmota/wiki/TasmoAdmin

It’s very simple to use...

RavenSystem, 2020-01-02:
v1.1.1 fixes bug to enter in setup mode toggling controls.

If you do it too quick, doesn't work.

Dirk23, 2020-01-02:
With 1.1.1 i can now enter Setup Mode By pressing 16x the Button for the Blind in HomeKit! Thnx @RavenSystem

RavenSystem, 2020-01-02:
@Dirk23 You can do it with 10 times. Try to pressing 10 times in 6-8 seconds.

closed 2020-01-10 by RavenSystem