RavenSystem/esp-homekit-devices #711: Feature Request: automatic updates at a certain time of day

tonysprenk, 2020-01-01:
It would be really nice to have a device search for a new update and update automatically at a certain time of day.

Dirk23, 2020-01-01:
I guess Auto OTA does what you want, but not at the time you want?

pilot1981, 2020-01-01:
To manually update I need to enter in setup mode first?

Dirk23, 2020-01-01:
> To manually update I need to enter in setup mode first?

Yes, and you Cam der

tonysprenk, 2020-01-02:
Auto OTA still requires the device to be put in setup mode. I’m talking about no user interaction, just a time of day trigger.

Dirk23, 2020-01-02:
I think it would make more sense to trigger the update procedure yourself. There are enough reports about something not working after updating to a new version. Then it would be better to test the update only on a less relevant device before you paralyze your whole house.
I would prefer the following 2 methods:
1. a procedure like pressing the button for 8 seconds, triggerable via HomeKit for all device types of HAA, which triggers the update manually. I have not yet been able to trigger the setup mode via HomeKit, for example with roller blinds.
2. a central management app/website that can trigger single updates or mass updates.

RavenSystem, 2020-01-02:
> I think it would make more sense to trigger the update procedure yourself. There are enough reports about something not working after updating to a new version. Then it would be better to test the update only on a less relevant device before you paralyze your whole house.

I totally agree.

> I would prefer the following 2 methods: ...

You can add an additional switch accessory with an action system to enter setup mode or update. Or even you can add 2 switches, one to enter setup mode and other to update. Then you can put those switch in other room called "Updates", or "Setup mode". Using HomeKit Ambients, you can trigger all of them at same time, even with a HomeKit automation you can update all of them at a certain time of the day.

To enable setup mode with window covering, you must stop your curtain 8 times. In other words, you must click HomeKit icon 16 times very quickly.

Dirk23, 2020-01-02:
@RavenSystem well, that sounds like a good plan. Unfortunately, I'm not a JSON professional and I haven't really gotten into this configuration stuff yet. What would be an example for these additional switches and what do you mean by Homekit Ambients?

Here is the JSON I use for my blinds so far

`{"c":{"l":13,"b":[{"g":10,"t":5}]},"a":[{"t":45,"o":18,"c":14,"f":70,"0":{"r":[{"g":12,"v":1},{"g":5}]},"1":{"r":[{"g":12},{"g":5,"v":1}]},"2":{"r":[{"g":12},{"g":5}]},"3":{"r":[{"g":12,"i":1},{"g":5}]},"4":{"r":[{"g":12},{"g":5,"i":1}]},"f0":[{"g":0}],"f1":[{"g":9}]}]}`

Kienz, 2020-01-02:
@RavenSystem What is "very quickly"? I can't get it to work. Tried it many times and setup mode is not entered.

Dirk23, 2020-01-02:
> @RavenSystem What is "very quickly"? I can't get it to work. Tried it many times and setup mode is not entered.

Same here.

RavenSystem, 2020-01-02:
I will check entering setup mode from curtains.

@Dirk23 If you enable "Auto OTA Updates" in setup, you only will need a switch to enter setup mode. What do you want? One switch with setup mode, or both (setup mode and update)?

"Ambients" means "Scenes". Sorry in Spanish is called "Ambientes".

Dirk23, 2020-01-02:
@RavenSystem I think 2 switches would be best, one for setup mode and one for update, because not every time I go into setup mode I want to have an update too.

RavenSystem, 2020-01-02:
@Dirk23 The "Auto OTA Updates" is enabled, update is done only if you enter in "setup mode" and you don't load setup web page from any browser during 90 seconds. If you load setup web page, your device will not be updated.

Dirk23, 2020-01-02:
Maybe we should leave open the possibility to have both :)

peros550, 2020-01-02:
Please consider an alternative way of triggering the update (if at all possible). 

The way it was implemented in Ravencore where you needed to use an alternative app like Eve was very clever :)

Adding more and more switches makes the management complicated. Also, some Siri commands (I.e “turn on all switches”) may trigger actions we don’t want to.

macjl, 2020-01-02:
I think an alternative would be to add an option for accessories to work in insecure mode (no HomeKit authentication), and not been displayed in HomeKit.

With that method, we could enter in setup mode by calling a specific URL. That would permit to a third party developer to build an app to manage HAA devices.

An other method I’m using with currents versions of HAA, is to add a kill switch on accessories I use for update and setup mode. It prevents to enter in setup mode accidentally. But it add many switches in Home. I have an idea for reducing number of switch that I will propose in another issue.

pilot1981, 2020-01-02:
> @RavenSystem What is "very quickly"? I can't get it to work. Tried it many times and setup mode is not entered.

Also me same issue

RavenSystem, 2020-01-02:
@Dirk23 JSON with 2 additional switches to enter setup and update:
`{"c":{"l":13,"b":[{"g":10,"t":5}]},"a":[{"t":45,"o":18,"c":14,"f":70,"0":{"r":[{"g":12,"v":1},{"g":5}]},"1":{"r":[{"g":12},{"g":5,"v":1}]},"2":{"r":[{"g":12},{"g":5}]},"3":{"r":[{"g":12,"i":1},{"g":5}]},"4":{"r":[{"g":12},{"g":5,"i":1}]},"f0":[{"g":0}],"f1":[{"g":9}]},{"1":{"s":[{"a":1}]}},{"1":{"s":[{"a":2}]}}]}`

RavenSystem, 2020-01-02:
Remember that you can group those switches to have less icons.

i3laze, 2020-01-03:
..., {"1":{"s":[{"a":1}]}},{"1":{"s":[{"a":2}]}}
So this is it? Looks like a good update for a Wiki.

Rocket200, 2020-01-04:
@RavenSystem can you help to make Update Button(s) for my Device ?
I chose only OTA Updates (no Auto Update)

Gosund SP111 V2:

{"c":{"l":2,"b":[{"g":13,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":15},{"g":0,"v":1}]},"1":{"r":[{"g":15,"v":1},{"g":0}]},"b":[{"g":13}]}]}

Energy rate is possible?
with Tasmota it works

nino1981be, 2020-01-04:
Hi,
I modified the json but I still do not see in eve the possibility of configuring or updating as with the old version ravencore 0.9.6 here is my json
{ "C": { "l": 13, "b": [{ "g" 0, "t": 5}]}, "a": [{ "0": { "r": [{ "g": 12}]}, "1": { "r": [{ "g": 12, "v": 1}]}, "b": [{ "g": 0}, { " g ': 14}, { "g": 14, "t": 0}]}]}

then I add what I thought was right to get there {"c": {"l": 13, "b": [{"g": 0, "t": 5}]}, "a" : [{ "0": { "r": [{ "g": 12}]}, "1": { "r": [{ "g": 12, "v": 1}]}, " b ": [{" g ": 0}, {" g ": 14}, {" g ": 14," t ": 0}]}]}, {" 1 ": {" s ": [{ "a": 1}]}}, { "1": { "s": [{ "a" 2}]}}]}

but that does not change anything I still do not have any bare configuration of possibility to update I am in haa 1.2
thanks for the help

closed 2020-01-10 by RavenSystem