RavenSystem/esp-homekit-devices #712: Unable to connect to the SONOFF wifi hotspot

Hornett69, 2020-01-02:
I have an issues with SONOFF Basic R2.
1- Flash SONOFF is ok with esptools.py
2- Connect to hotspot wifi and configuration with haa reporsitory
3- after 30 minutes no hotspot HAA-XXXXXX
4- I unplug the device and plug it to the electrical sector. After, I can see the HAA-XXXXXX hotspot but cannot connect and it disappears after a few minutes ... I tried several times but without success... I don't understand why !

Does someone have an idea to help me ?

pawelosin, 2020-01-02:
Check this out - https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#entering-setup-mode

Hornett69, 2020-01-02:
Thank you... I did not know that the method had changed ... It is solved ... Perfect !
Regards.

closed 2020-01-02 by Hornett69