RavenSystem/esp-homekit-devices #713: Kill switches as Characteristic of accessory, instead of separate switches

macjl, 2020-01-02:
Hello José,

I'l not an HomeKit expert, but do you think that kill switch could be grouped with the accessory it is controlling? Since iOS 13.2, user can choose if they want to group or not the devices that are in the same accessory. The benefits would be less icon in Home app.

With HAA, the kill switches are separate accessory from the ones they controls. By grouping the characteristic of the kill switch within the same accessory, I think it will permit grouping them.

Do you know if it's doable?

RavenSystem, 2020-01-02:
You could separate them, but you could not move to a different room. And would be necessary to recode all "kill switches" subsystem.

RavenSystem, 2020-01-02:
Remember that you can group "kill switches" (o any other switch) from different accessories.

macjl, 2020-01-02:
Manually grouping accessories don't work the same way. If you group multiple switches, by activating the group, all switches will be triggered at the same time.
When an accessory contain different characteristics, if you click on the accessory in Home, only the principal characteristic will be triggered. For the others, you have to open the accessory details. I think it's much better for the use case of kill switches.

macjl, 2020-01-02:
> You could separate them, but you could not move to a different room.

I didn’t noticed that. Thank you for pointing it as it would be a regression for some users.

closed 2020-01-02 by macjl

pilot1981, 2020-01-02:
You can use specific room...
Kill switch what means? To enter in setup mode? Which difference there is with dummy switch?

macjl, 2020-01-02:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#Kill-Switch