RavenSystem/esp-homekit-devices #714: Shelly 1 switch as electric door mechanism trigger

morphix9, 2020-01-02:
Shelly 1 
action: holding the relay open for 5 seconds
default state is closed/locked

dominikkosciolek, 2020-01-09:
{"c":{"b":[{"g":5,"p":0,"i":1,"t":5}]},
"a":[
{
"t": 4,
"0":{"r":[{"g":4,"v":1}]},
"1":{"r":[{"g":4}]},
"i": 5
"b":[{"g":5}]
}
]}

RavenSystem, 2020-01-10:
`{"c":{"b":[{"g":5,"p":0,"i":1,"t":5}]},"a":[{"t": 4,"i":5,"0":{"r":[{"g":4,"v":1}]},"1":{"r":[{"g":4}]},"b":[{"g":5}]}]}`

closed 2020-01-10 by RavenSystem

morphix9, 2020-01-17:
Works like a charm, thanks!

lickonline, 2020-10-25:
looked also for a json for a electric door mechanism
this json works perfect, thank you (all)!
_______________________________________
I changed to a triggering time of 2.5s, because 5s is a bit too long for my needs
I changed to a reboot state "locked", because I don*t want to open the door, when rebooting

json with the two small modifications:
{"c":{"b":[{"g":5,"p":0,"i":1,"t":5}]},"a":[{"0":{"r":[{"g":4,"v":1}]},"1":{"r":[{"g":4}]},"t":4,"i":2.5,"b":[{"g":5}],"s":1}]}