RavenSystem/esp-homekit-devices #716: Upgrade RavenCore 1.2 to HAA

shimon88, 2020-01-03:
Hi all,

I have issues with upgrading one of my Sonoff mini's to HAA.
I have followed all guides I could find here with no avail.

when I set in EVE "HAA firmware" and then "firmware update", the device is rebooting and then becomes online in home app. so no upgrade is taking place.

It is located in a place hard to reach to do a fresh manual upgrade.

any idea?
thanks

i3laze, 2020-01-03:
Actually, RavenCore 1.3 is available. Looks like does the single thing - correct upgrade to HAA.

Please try just RC upgrade (to 1.3) and tell us how it went.

shimon88, 2020-01-03:
Thank you. 
Just tried an update via the eve app but no change. It looks like it just doing a reboot and then go back online.

GPL71, 2020-01-03:
The update turn you in Ravencore 1.3, and an automatic system turn you in HAA.
Now you need digit the device IP and insert in the configuration the json

shimon88, 2020-01-03:
I can get it to update to 1.3 either. It only reboots. Maybe I’m doing something wrong

shimon88, 2020-01-05:
Ok so I figured it out.
The ip address was blocked in the router's firewall.
removed and all worked

thanks for all for trying to help

closed 2020-01-05 by shimon88