RavenSystem/esp-homekit-devices #718: HAA request for shelly 2.5 window covering

kx3kx3kx3, 2020-01-04:
please help me with  json string for shelly 2,5 window coveering with up/down button

dido92, 2020-01-05:
Please I need the same json string. I have a big problem whit my Shelly 2.5. It’s very important. Please

dominikkosciolek, 2020-01-05:
I have the same problem, this JSON works but witout stop after second push button

{"c":
{"l":0,"b":[{"g":2,"t":5}]},
"a":[{
"t":45,
"o":30,
"c":30,
"0":{"r":[{"g":15},{"g":4,"v":1}]},
"1":{"r":[{"g":15,"v":1},{"g":4}]},
"2":{"r":[{"g":15},{"g":4}]},
"3":{"r":[{"g":15},{"g":4,"i":1}]},
"4":{"r":[{"g":15,"i":1},{"g":4}]},
"f0":[{"g":5,"p":0}],
"f1":[{"g":13,"p":0}],
"f2":[{"g":5 ,"p":0 },{"g":13 ,"p":0 }]}]}

GPL71, 2020-01-06:
Guys use the sample json

RavenSystem, 2020-01-06:
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#shelly25-window-covering-driver-with-external-push-buttons-status-led-and-built-in-setup-button-opening-time-18-seconds-closing-time-14-seconds-non-lineal-correction-70

dido92, 2020-01-07:
> https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#shelly25-window-covering-driver-with-external-push-buttons-status-led-and-built-in-setup-button-opening-time-18-seconds-closing-time-14-seconds-non-lineal-correction-70

Thanks but this exemplar it doesn’t work for me 😩

dominikkosciolek, 2020-01-07:
Please use this template 

{"c":
{"l":0,"b":[{"g":2,"t":5}]},
"a":[{
"t":45,
"o":30,
"c":30,
"0":{"r":[{"g":15},{"g":4,"v":1}]},
"1":{"r":[{"g":15,"v":1},{"g":4}]},
"2":{"r":[{"g":15},{"g":4}]},
"3":{"r":[{"g":15},{"g":4,"i":1}]},
"4":{"r":[{"g":15,"i":1},{"g":4}]},
"f3":[{"g":5,"p":0}],
"f4":[{"g":13,"p":0}]}
]}

RavenSystem, 2020-01-07:
He said that doesn’t work for him. And template from examples is better because uses less memory.

closed 2020-01-07 by RavenSystem

raddi, 2020-01-09:
+1 I have Windows covering which triggers with Push Button. There are 2 Toggle-Buttons: Up and Down. If you hold one button for 2 sek (and let go after that), it will start to do the motion. To stop you have to press the opposite button. So to make this kind of window covering, there have to be 2 stop actions. Stop after closing, stop after opening. Without the stop action, this should work:

{"c":
{"l":0,"b":[{"g":2,"t":5}]},
"a":[{
"t":45,
"o":25,
"c":25,
"0":{"r":[{"g":15},{"g":4,"v":1,"i":2}]},
"1":{"r":[{"g":15,"v":1,"i":2},{"g":4}]},
"f0":[{"g":5,"p":0}],
"f1":[{"g":13,"p":0}],

Opening 50% etc. is not possible with this config.

RavenSystem, 2020-01-09:
@raddi Using f0 and f1 with push buttons is not a good idea.

Better use this template for up and down push buttons:
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#shelly25-window-covering-driver-with-external-push-buttons-status-led-and-built-in-setup-button-opening-time-18-seconds-closing-time-14-seconds-non-lineal-correction-70

raddi, 2020-01-09:
@RavenSystem The problem with this template is, that i need inching for the relais because the window covering is working like this: Press button for 2 seconds: window cover will close or open completely.

RavenSystem, 2020-01-09:
@raddi You press button to start moving, and press again same button to stop.

raddi, 2020-01-09:
Okay let my try to explain my situation:

I have a wall switch with 2 Push Buttons, one for up, one for down. If i press "up" for 2 sec, the window cover will start to roll up, i don't need to hold the up button for that, the cover will automatically rise till its open. When its up, i press "down" for 2 sec and it will start do go down completely. When its rolling up or down, i can press the opposite button to stop the motion.

Thats the normal behavior of most curtain controllers - i think.

Now i try to connect the shelly2.5 in between. So the relay of the shelly needs to have the inching function for 2 sec. I press the up/down button, the relay will close the circuit up/down for 2s and let go. Now i need two stop functions. If its rolling down, the relay has to trigger the up relay shortly and when its rolling up, it needs to trigger the down relay shortly. 

https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#shelly25-window-covering-driver-with-external-push-buttons-status-led-and-built-in-setup-button-opening-time-18-seconds-closing-time-14-seconds-non-lineal-correction-70

This template has no inching, so the relay is closed the whole closing or opening time?

I hope you understand my situation :) Thank you for your fast response!

RavenSystem, 2020-01-09:
@raddi Yours is not the typical case. Your window cover mechanism is not the standard. HAA only has one stop action, and as you say, you need 2 stop actions: stops while opening and stop while closing. I will implement them in a new version.

raddi, 2020-01-09:
@RavenSystem thank you so much! :)

michapi, 2020-05-20:
> Okay let my try to explain my situation:
>
> I have a wall switch with 2 Push Buttons, one for up, one for down. If i press "up" for 2 sec, the window cover will start to roll up, i don't need to hold [...]

Hey, is this problem solved? Do you have working configuration?