RavenSystem/esp-homekit-devices #719: Shelly 2.5 Windows Covering - need help!!

dido92, 2020-01-05:
Hi guys. Sorry but i need your help. i need this config in HAA. Can you help me??

board type: 5
device type: 12
external toogle : 1 
time close: 15
time open: 15 
THANKS

GPL71, 2020-01-06:
@dido92 check json sample page.... replace time there with your correct time

dido92, 2020-01-06:
> @dido92 check json sample page.... replace time there with your correct time

It doesn’t work for me! Because the json sample it’s for external switch, I have only button. 😩 please help

RavenSystem, 2020-01-06:
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#shelly25-window-covering-driver-with-external-push-buttons-status-led-and-built-in-setup-button-opening-time-18-seconds-closing-time-14-seconds-non-lineal-correction-70

RavenSystem, 2020-01-06:
Duplicated: https://github.com/RavenSystem/esp-homekit-devices/issues/718

closed 2020-01-06 by RavenSystem

dido92, 2020-01-07:
> @dido92 check json sample page.... replace time there with your correct time

It doesn’t work for me! Because the json sample it’s for who have switch, I have only button. 

> https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#shelly25-window-covering-driver-with-external-push-buttons-status-led-and-built-in-setup-button-opening-time-18-seconds-closing-time-14-seconds-non-lineal-correction-70

Thanks! But this example it doesn’t work for me 😩