RavenSystem/esp-homekit-devices #721: Sonoff 4ch Pro R2 Locks

asbesto01, 2020-01-08:
Ciao to everybody,

quick question. I have this json for a sonoff 4ch pro r2 with two locks and two switches:

{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"t":4,"i":0.4,"0":{"r":[{"g":12,"v":1}]},"1":{"r":[{"g":12,"v":0}]},"b":[{"g":0}]},{"t":4,"i":0.4,"0":{"r":[{"g":5,"v":1}]},"1":{"r":[{"g":5,"v":0}]},"b":[{"g":9}]},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":10}]},{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14}]}]}

I need to have 4 identical locks, with inching 0.4 

Can someone help me?

Thanks!

dominikkosciolek, 2020-01-09:
{"c":
{"l":13,"i":1,"b":[{"g":0,"t":5}]},
"a":[
{
"t":4,
"i":0.4,
"0":{"r":[{"g":12,"v":1}]},
"1":{"r":[{"g":12,"v":0}]},
"b":[{"g":0}]
},
{
"t":4,
"i":0.4,
"0":{"r":[{"g":5,"v":1}]},
"1":{"r":[{"g":5,"v":0}]},
"b":[{"g":9}]
},
{
"t":4,
"i":0.4,
"0":{"r":[{"g":4,"v":1}]},
"1":{"r":[{"g":4,"v":0}]},
"b":[{"g":10}]
},
{
"t":4,
"i":0.4,
"0":{"r":[{"g":15,"v":1}]},
"1":{"r":[{"g":15,"v":0}]},
"b":[{"g":14}]
}
]}

RavenSystem, 2020-01-09:
This approach is better, because using an inching too short with the accessory can cause issues with HomeKit. In this case, inching is better to use with relay:

```json
{"c":
{"l":13,"i":1,"b":[{"g":0,"t":5}]},
"a":[
{
"t":4,
"i":5,
"0":{"r":[{"g":12,"v":1,"i":0.4}]},
"1":{"r":[{"g":12}]},
"b":[{"g":0}]
},
{
"t":4,
"i":5,
"0":{"r":[{"g":5,"v":1,"i":0.4}]},
"1":{"r":[{"g":5}]},
"b":[{"g":9}]
},
{
"t":4,
"i":5,
"0":{"r":[{"g":4,"v":1,"i":0.4}]},
"1":{"r":[{"g":4}]},
"b":[{"g":10}]
},
{
"t":4,
"i":5,
"0":{"r":[{"g":15,"v":1,"i":0.4}]},
"1":{"r":[{"g":15}]},
"b":[{"g":14}]
}
]}
```

asbesto01, 2020-01-09:
Ciao Jose,

it's not clear to me which is the difference between your proposal and dominik one...

BTW, happy new year and thanks for the wonderful HAA you have created for us!!!!

Best regards

RavenSystem, 2020-01-09:
With Dominik JSON, lock icon opens and closes automatically after 0.4 seconds. If you say "Hey Siri, open lock", lock will be opened, but closed too quickly, and when HomeKit check lock status, it will be closed, and Siri tells you that she can not open it.

With mine, lock icon opens and closes automatically after 5 seconds, but relay works only 0.4 seconds to open your electrolock.

asbesto01, 2020-01-09:
thank you so much ;) tried and worked.. It can be closed!

Best regards

closed 2020-01-09 by RavenSystem

asbesto01, 2020-11-15:
need this to be reopened; I need a json for a SONOFF DUAL R2 with 2 locks.

both locks need 0.4 seconds inching, and 5 seconds opening time. Can you please help me?

Thanks!