RavenSystem/esp-homekit-devices #722: Sonoff T2 EU 3 - button + window covering

quartist, 2020-01-08:
Hello,

Im looking for json for Sonoff T2 EU 3 in configuration like 
button + window covering: 
normal switch [stable](button1 turn on/off lamp) |  open window cover [inching with interlock with button 3] |  close window cover [inching with interlock with button 2]

It is possible to build this in HAA?

RavenSystem, 2020-01-09:
Yes, it is possible.

RavenSystem, 2020-01-09:
Try this, but I don't know if GPIOs are right:

```json
{"c":{"l":13,"b":[{"g":10,"t":5}]},"a":[{"t":45,"o":18,"c":14,"f":70,"0":{"r":[{"g":12,"v":1},{"g":5}]},"1":{"r":[{"g":12},{"g":5,"v":1}]},"2":{"r":[{"g":12},{"g":5}]},"3":{"r":[{"g":12,"i":1},{"g":5}]},"4":{"r":[{"g":12},{"g":5,"i":1}]},"f3":[{"g":0}],"f4":[{"g":9}]},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":10}]}]}
```

closed 2020-01-10 by RavenSystem

quartist, 2020-01-19:
Hi,

tested! This working good. Left and middle are working good for window covering with interlock, right working like normal button. How to change time for "open" and "close"? In parameter "o":18 and "c":14 ?

RavenSystem, 2020-01-19:
> How to change time for "open" and "close"? In parameter "o":18 and "c":14 ?

Yes. "o": open , and "c": close, in seconds. You can use decimals ("o":12.7).

pilot1981, 2020-01-19:
@quartist I’m interested in your implementation: please, can you share more details and some pictures?

quartist, 2020-01-20:
@pilot1981 
Are you from PL?
What information/details do you need? Sonnof's are not mounted on wall yet. The house is being build.

pilot1981, 2020-01-20:
italy.
with electric/mechanical tool your are using with your windows? only some picture please

I would implement automated swing windows...

quartist, 2020-01-20:
ok, so...
I buyed anti-burglar blind which including electrical motor to closing and opening. Each blind have one 3-wires electrical cable (UP-brown, DOWN-black, BLUE-neutral). In this configuration left and middle buttons on sonoff T2EU3 are for control. One for UP - brown cable, Second for DOWN - black cable. Neutral connect with other neutral in box. I will taking couple of pictures ASAP. Maybe some video for YT.

pilot1981, 2020-01-21:
How much?