RavenSystem/esp-homekit-devices #723: sonoff basic toujours en mode configuration

clint80, 2020-01-09:
bonjour
j,ai flasher mon sonff basic avec la methode OTA
apres avoir configurer mon Json puis sauvegarder 
mon sonoff clignote une fois puis 8s apres clignote 5 fois
avez vous deja eu ce probleme 
merci

RavenSystem, 2020-01-10:
English, please.

clint80, 2020-01-10:
hello I  have flash my sonoff basic with the OTA method
 after having configured my Json then save 
 mon sonoff flashes once then 8s after flashes 5 times 
you have already had this problem 
thank you

RavenSystem, 2020-01-10:
Can you add them to HomeKit?

clint80, 2020-01-10:
in HomeKit, I see the HAA-XXXXXX icon
I type the code, it tries to synchronize it, but each time it fails

RavenSystem, 2020-01-10:
What JSON are you using?

clint80, 2020-01-10:
this is the model that is on your site: https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":14,"t":0}],"f4":[{"g":14}],"f2":[{"g":4,"t":0}],"f5":[{"g":4}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"10":{"a":0}},{"i":1,"1":{"m":[{"g":1,"v":2}]}}]}

clint80, 2020-01-10:
I was wrong for the sonoff basic, I took this one
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}

clint80, 2020-01-10:
after several flashes
I have successfully synchronized with HomeKit
thank you for all the work you do

closed 2020-01-10 by clint80