RavenSystem/esp-homekit-devices #727: Auto-close Garage door

calma22, 2020-01-09:
Hello to everybody,

i have a quick question. I have a sonoff 4ch pro r2 with two garage opener (one for Box, one for car gate), i would like to know if it's possible to set one auto-command (to the same output) for close the gate after XX seconds without click again on my phone.

Can someone help me?

Thanks!

RavenSystem, 2020-01-09:
Yes, you can. Post your JSON.

calma22, 2020-01-16:
Hi sorry for delay!! 
Here we are my JSON:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":0.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":0.5}]},"3":{"a":2},"t":40,"d":18,"e":5,"s":1,"b":[]},{"0":{"r":[{"g":5,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":5,"i":0.5},{"g":5,"v":1,"i":0.5},{"g":5,"v":1,"i":0.5}]},"3":{"a":2},"t":40,"d":18,"e":5,"s":1,"b":[]},{"0":{"r":[{"g":4,"v":1,"i":0.2}]},"1":{"r":[{"g":4,"v":0,"i":0.2}]},"t":1,"b":[],"s":0},{"0":{"r":[{"g":15,"v":1,"i":0.2}]},"1":{"r":[{"g":15,"v":0,"i":0.19}]},"t":1,"b":[],"s":0}]}

Thanks a lot for your help

calma22, 2020-01-24:
Hi Raven,
did you find any solution for this? 

Thanks

calma22, 2020-02-03:
> Yes, you can. Post your JSON.
> […](#)

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":0.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":0.5}]},"3":{"a":2},"t":40,"d":18,"e":5,"s":1,"b":[]},{"0":{"r":[{"g":5,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":5,"i":0.5},{"g":5,"v":1,"i":0.5},{"g":5,"v":1,"i":0.5}]},"3":{"a":2},"t":40,"d":18,"e":5,"s":1,"b":[]},{"0":{"r":[{"g":4,"v":1,"i":0.2}]},"1":{"r":[{"g":4,"v":0,"i":0.2}]},"t":1,"b":[],"s":0},{"0":{"r":[{"g":15,"v":1,"i":0.2}]},"1":{"r":[{"g":15,"v":0,"i":0.19}]},"t":1,"b":[],"s":0}]}

dominikkosciolek, 2020-02-06:
Any info?

RavenSystem, 2020-02-07:
You must use a third party HomeKit client (ex: EVE App) to configure auto-close timers:

`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"d":3600,"0":{"m":[{"g":2,"v":1}]}},{"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":0.5},{"g":12,"v":1,"i":0.5}]},"3":{"a":2},"t":40,"d":18,"e":5,"5":{"m":[{"g":1}]},"6":{"m":[{"g":1,"v":1}]}},{"d":3600,"0":{"m":[{"g":4,"v":1}]}},
{"0":{"r":[{"g":5,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":5,"i":0.5},{"g":5,"v":1,"i":0.5}]},"3":{"a":2},"t":40,"d":18,"e":5,"5":{"m":[{"g":3}]},"6":{"m":[{"g":3,"v":1}]}},{"0":{"r":[{"g":4,"v":1,"i":0.2}]},"1":{"r":[{"g":4,"i":0.2}]}},{"0":{"r":[{"g":15,"v":1,"i":0.2}]},"1":{"r":[{"g":15,"i":0.19}]}}]}`

closed 2020-02-07 by RavenSystem

dominikkosciolek, 2020-02-10:
it is not necessary to use an external client, below clearer version autoclose for Sonoff SV 
Autoclose time 2 minutes

{"c":{"l":13,"b":[{"g":0,"t":5}]},
"a":[
{"d":120,
"0":{"m":[{"g":2,"v":1}]}
},
{"t":40,"d":16,"c":16,
"0":{"r":[{"g":12,"v":1,"i":0.6}]},
"1":{"a":0},
"2":{"r":[{"g":12,"i":0.6},{"g":12,"v":1,"i":0.6},{"g":12,"v":1,"i":0.6}]},
"3":{"a":2},
"5":{"m":[{"g":1}]},
"6":{"m":[{"g":1,"v":1}]},
"f3":[{"g":14,"t":0}],
"f4":[{"g":14}],
"f2":[{"g":4,"t":0}],
"f5":[{"g":4}]
}
]}